FreeCAD/FreeCAD #11642: [Feature Request] Support for advanced material models offered by CalculiX in FEM

FEA-eng, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Perhaps I should split this one into multiple issues since the support for each material model can be considered separately but I don't want to flood the repository with too many FRs. Anyway, here are the advanced material models that could be added:
- orthotropic and anisotropic linear elasticity
- hyperelasticity (several models available) and hyperfoam
- creep
- kinematic, combined and Johnson-Cook hardening for plasticity
- Mohr-Coulomb plasticity
- tension-only and compression-only materials

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-02-29:
is this compatible with the material overhaul?

FEA-eng, 2024-02-29:
> is this compatible with the material overhaul?

Currently, the only nonlinear material model available in FEM for CalculiX uses a separate [FEM MaterialMechanicalNonlinear](https://wiki.freecad.org/FEM_MaterialMechanicalNonlinear) object independent of the material editor. The new PR updates it. Hopefully, in the future more advanced material models will be added either to this object or to the editor itself.

maxwxyz, 2024-03-01:
@davesrocketshop FYI for the _to think about_ list

davesrocketshop, 2024-03-01:
The new material system is designed to handle cases such as this. I have some "theoretical" non-linear models I've used for testing.

Any pointers to documentation or example models I could reference?

davesrocketshop, 2024-03-01:
> Currently, the only nonlinear material model available in FEM for CalculiX uses a separate [FEM MaterialMechanicalNonlinear](https://wiki.freecad.org/FEM_MaterialMechanicalNonlinear) object [...]

This is an example of the 2DArray property. Interpolation isn't in there yet because of priorities but it will be easy to implement.

davesrocketshop, 2024-03-01:
Also, there are some example non-linear models in the repository:
src/Mod/Material/Resources/Models/Mechanical/OrthotropicLinearElastic.yml
src/Mod/Material/Resources/Models/Mechanical/OgdenYld2004p18.yml

These aren't the only way to implement those models, but they should give you some ideas. I actually talked to a materials researcher to make these models as flexible as possible. I can help with the design if you know what you're trying to achieve.

FEA-eng, 2024-03-01:
@davesrocketshop There are 3 main aspects to consider (assuming that we are talking only about CalculiX since Elmer doesn't even have plasticity):
**1) Defining properties of advanced material models** - I would count on your new material editor here
**2) Applying them to FEM models** - currently done with _FEM MaterialSolid_ (for standard properties) and _FEM MaterialMechanicalNonlinear_ (for plasticity), I don't have much experience with coding here but I'll try to help a bit and @marioalexis84 might be able to help way more
**3) Translating them to the input file syntax** - I can help with that since I'm familiar with the CalculiX .inp file syntax

I would start with orthotropy (requested several times on the forum). It's relatively simple to add but requires the possibility to define material orientations (mentioned in the issue description). Hyperelasticity (multiple models) would also be very useful. The rest could wait for now since they are rarely needed.

FEA-eng, 2024-03-01:
> I'm not sure what you mean for hyperelasticity, but there is also a 3DArray capability. So for example, MaterialMechanicalNonlinear at different temperatures. Again interpolation is not in there yet, but doable. And it will get in there at some point depending on priorities.

Right, there’s also temperature dependence. It can be used for different material models. In CalculiX, it’s defined by specifying material contants at different selected temperatures and the solver does the interpolation and extrapolation internaly.

Hyperelasticity is what you’ve mentioned as Ogden model plus multiple other models (Yeoh, Marlow, Neo-Hookean and so on). It’s defined by several constants depending on the model. Of course, I can always provide examples if needed.

davesrocketshop, 2024-03-01:
Just let me know what you need, and I'll help get it done!

FEA-eng, 2024-03-01:
> Just let me know what you need, and I'll help get it done!

For the most basic definition of orthotropic linear elasticity we only need this:
`E1, E2, E3, ν12, ν13, ν23, G12, G13, G23`

And I see that it's already in your code. But CalculiX accepts also this form of definition:
`D1111, D1122, D2222, D1133, D2233, D3333, D1212, D1313, D2323`

Those are elastic stiffness matrix terms but this type of definition is much less common.

Then anisotropic linear elasticity requires this:
`D1111, D1122, D2222, D1133, D2233, D3333, D1112, D2212, D3312, D1212, D1113, D2213, D3313, D1213, D1313, D1123, D2223, D3323, D1223, D1323, D2323`

I guess that we could skip it for now too.

So we could try to add orthotropy based on the existing parameters and think about hyperelasticity for which we need multiple models (should be easily selectable from a list):
- Arruda-Boyce: 
`µ, λm, D`
- Mooney-Rivlin:
`C10, C01, D1`
- Neo-Hooke:
`C10, D1`
- Ogden with N=1:
`µ1, α1, D1`
- Ogden with N=2:
`µ1, α1, µ2, α2, D1, D2`
- Ogden with N=3:
`µ1, α1, µ2, α2, µ3, α3, D1, D2, D3`
- Polynomial with N=1:
`C10, C01, D1`
- Polynomial with N=2:
`C10, C01, C20, C11, C02, D1, D2`
- Polynomial with N=3:
`C10, C01, C20, C11, C02, C30, C21, C12, C03, D1, D2, D3`
- Reduced Polynomial with N=1:
`C10, D1`
- Reduced Polynomial with N=2:
`C10, C20, D1, D2`
- Reduced Polynomial with N=3:
`C10, C20, C30, D1, D2, D3`
- Yeoh:
`C10, C20, C30, D1, D2, D3`

We may not need all of them at once but at least the basic ones like the first 3 plus the last one (and maybe also Ogden) would be good to have for now.

davesrocketshop, 2024-03-01:
Give me a bit and I'll have something for you to review.

davesrocketshop, 2024-03-02:
Looking into this more brings up many questions, including data types, documentation, etc. Would you have time for a zoom call? Or a chat on discord? This would be a lot quicker in an interactive format

marioalexis84, 2024-03-02:
Just note that in the Material data there should not be anything that refers to something specific to a particular solver. The data must be presented in a general way and then each solver must decide how to deal with that data.

FEA-eng, 2024-03-02:
> Looking into this more brings up many questions, including data types, documentation, etc. Would you have time for a zoom call? Or a chat on discord? This would be a lot quicker in an interactive format

Ok, I'll dm you on the forum.