FreeCAD/FreeCAD #11644: [Feature Request] Spring elements for CalculiX FEM

FEA-eng, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is [FEM ConstraintSpring](https://wiki.freecad.org/FEM_ConstraintSpring) but it only works with Elmer. CalculiX also offers spring elements (3 types actually) and they are very useful so it would be good to support them.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

3xxx, 2026-08-16:
Really need this feature, looking forward to it being available soon.

luzpaz, 2026-08-30:
@3xxx please bring this up on the FEM subforum. Perhaps a dev or 2 would help make this happen faster?

FEA-eng, 2026-08-30:
> [@3xxx](https://github.com/3xxx) please bring this up on the FEM subforum.

He did that: https://forum.freecad.org/viewtopic.php?t=107024

> Perhaps a dev or 2 would help make this happen faster?

Would be great to add this, but it would have to be implemented in a very different way than what the existing Spring constraint does using Elmer. @marioalexis84 can correct me if I’m wrong, but from what I know, Elmer simply has a dedicated BC type where you just specify the face and the stiffness values. In CalculiX, you have to define spring elements explicitly for each node. I will add more details to the aforementioned forum thread.

marioalexis84, 2026-08-31:
Exactly. The spring object is a boundary condition for the Elmer solver. In CalculiX, it must be defined as an element type.
However, Elmer offers another way to define springs on nodes: https://www.nic.funet.fi/index/elmer/doc/ElmerModelsManual.pdf#chapter.11