FreeCAD/FreeCAD #11645: [Feature Request] Hydrostatic (non-uniform) pressure load in FEM

FEA-eng, 2023-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hydrostatic pressure loads are commonly used in pressure vessel analyses. Unfortunately, CalculiX doesn't support them in the form of keywords (only as a dload.f subroutine) but there are ways to define them in preprocessors - internally, they could be converted to multiple *CLOAD or *DLOAD lines.

Support for general non-uniform pressure loads (e.g. defined by a user-specified equation) would be great but hydrostatic pressure will be sufficient for now.

Old forum discussion: https://forum.freecad.org/viewtopic.php?t=37631

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jwharington, 2025-11-11:
Yes happy to

3xxx, 2026-08-16:
Really need this feature, looking forward to it being available soon.