FreeCAD/FreeCAD #11659: DXF export option mis-placed in "Import options" section

kpemartin, 2023-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The DXF page of Preferences shows the "Treat ellipses and splines as polylines", which is an export option, in the "Import options" group rather than in the "Export options" group

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
and also the main development branch
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kpemartin, 2023-12-05:
I can fix this while I re-word the option label for color importing (#8806).

How do I add a tag/label to an issue? This one should be "File format: DXF"

kpemartin, 2023-12-05:
I changed this file (for another issue) and found it had no effect until I forcibly re-built the Draft project (this is in Visual Studio).
It would appear that this project has no dependency on the preferences-dxf.ui file. Is this a flaw in the CMake config, or is this just something that can't be made to work in VS?

kpemartin, 2023-12-05:
How are the translations managed? I see (for instance) the translations/Draft_fr.ts file contains French translations for the messages I've changed, but it also contains both the file and line number of the original message but also the original (English) message text. Is this a generated file? Who provides the translations? Does the <source> have to match the text in the preferences-dxf.ui file for the translation to work? Have I missed a file that answers most of these questions?

Roy-043, 2023-12-06:
Qt has a dialog designer. But you can also edit the XML. At least I have done that when working on the Draft preferences.

The UI files are compiled into Draft_rc.py. This file is references at runtime.
To test your file while you are working on it, you can temporarily change the path in InitGui.py.
`FreeCADGui.addPreferencePage(":/ui/preferences-dxf.ui", QT_TRANSLATE_NOOP("QObject", "Import-Export"))`
To for example:
`FreeCADGui.addPreferencePage("D:/test/preferences-dxf.ui", QT_TRANSLATE_NOOP("QObject", "Import-Export"))`
After changing InitGui.py you have to restart FreeCAD. But to see the updated UI all you have to do is restart the Preferences Editor.

The TS files are generated by scripts. @yorikvanhavre handles this. If you change the English text the new TS file will contain that string for translation. No further action is required for UI files. For other files see https://wiki.freecad.org/Translating_an_external_workbench.

PR #11938 (2024-01-11): Move the "Treat ellipses...as polylines" to DXF *output* options group
This option control DXF export but was confusingly in the Import options area. Fixes #11659
commit d52cf81b (2024-01-11): Move the "Treat ellipses and splines as polylines" to DXF *output* options
This option control DXF export but was confusingly in the Import options area. Fixes #11659
closed 2024-01-11 by yorikvanhavre