FreeCAD/FreeCAD #11661: [Feature request] Sketcher: highlight constrained entities when a constrain is selected

pierreporte, 2023-12-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is inspired by a discussion with @M4xTheMan about #11603. The idea is that it can be difficult to identify which entities are constrained by which constraint. It would be useful if, when a constrain is selected on the 3D view or task panel, the geometry that is constrained by it is highlighted.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35249 (Git) AppImage
Build type: Release
Branch: main
Hash: 94ca51b7799b2ac60a2a56d7e6e753bb97a73671
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * Render 2023.8.18
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2023-12-06:
@Roy-043 I didn’t know this tool. For me it solves the issue. However it may be wanted to have an automatic highlight. @M4xTheMan what do you think?

M4xTheMan, 2024-04-12:
@pierreporte Sorry, it took me a while. Unfortunately, you haven't linked to the thread you've been talking about. Could you please add it to refresh my memory?

pierreporte, 2024-04-12:
@M4xTheMan What forum post? It isn’t mentioned here, only in #11603 where it is written that I was unsure of its existence. If I was unable to find the information at the time, I am unlikely to succeed now.

M4xTheMan, 2024-04-15:
Never mind, I've mixed it up.

> @Roy-043 I didn’t know this tool. For me it solves the issue. However it may be wanted to have an automatic highlight. @M4xTheMan what do you think?

I'm fine with closing this ticket.