FreeCAD/FreeCAD #11665: [Problem] FEM legend may still be partially covered under some screen resolutions

FEA-eng, 2023-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PR #10552 submitted by @Syres916 significantly improved the FEM legend - now it's much less likely to be partially covered by the navigation cube and the labels are more visible (plus there are settings for their font in Preferences). However, there are still 2 small issues mentioned in this comment: https://github.com/FreeCAD/FreeCAD/pull/10552#issuecomment-1729950881

So under some screen resolutions, the legend can still get partially covered by the navigation cube. Also, larger label font sizes may lead to values going beyond the screen.

It should be possible to fix it but if it turns out to be too troublesome, my idea is to just move the legend to the left side of the screen (as shown in this comment: https://github.com/FreeCAD/FreeCAD/pull/10552#issuecomment-1735549319).  Any potential issues with this approach ?

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-06:
Morning @FEA-eng the line that moves the FEM Legend is https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/SoFCColorBar.cpp#L139 the concern that needs testing is the Overlay (which I don't use) as I don't known if the screen area includes that underneath the transparent tree. You're welcome to give it a try with I guess 1.2 instead of 4.95, compile, test and amend from there.

FEA-eng, 2023-12-06:
@Syres916

Hi,
thanks for the reply and the information. I'll give it a try. But do you think that it makes sense to try repairing the aforementioned issues while keeping the legend on the right side of the screen ? And can you try doing it ? Or does the idea of moving the legend to the left seem to be a better approach ?

FEA-eng, 2023-12-06:
@Syres916 What if we just add an option in Preferences to move the legend to the left or right ? Or maybe even make it possible to change this position parameter ? That should resolve those issues in most cases. The only question would be what setting to use as default.

Syres916, 2023-12-06:
The best I can say is when I get chance to look at the feedback OpenBrain gave me regarding the legend text colour and size updating in Preferences then it would be the best time to add this also but like I said elsewhere it's going to be Xmas hols at the earliest.

Syres916, 2023-12-27:
@FEA-eng as promised this will be WIP tomorrow and as long as can understand Openbrain's explanation I should be ready for you to test a branch by Friday evening.

FEA-eng, 2023-12-27:
@Syres916 Glad to hear that. It would be great to have this issue fully resolved soon.

Syres916, 2023-12-28:
@FEA-eng I'm afraid it's as I was fearing that I just cannot make any progress, my skill-set is not up to doing this,.
You're going to have to raise it to another developer, sorry but 8 hours completely wasted.

FEA-eng, 2023-12-28:
> @FEA-eng I'm afraid it's as I was fearing that I just cannot make any progress, my skill-set is not up to doing this,. You're going to have to raise it to another developer, sorry but 8 hours completely wasted.

Then I would just go with plan B and either move the legend to the left or (better) add a preference to position it on the left or right or even specify the exact location using this parameter that is currently set to 4.95. This should be doable, right ?

luzpaz, 2024-09-09:
The related #12742 could also help free up more 'screen real-estate' for smaller screens

luzpaz, 2024-10-24:
Is this issue still relevant ?

FEA-eng, 2024-10-24:
> Is this issue still relevant ?

Sure, issues with the the color legend bar’s position are still one of the most annoying and difficult to fix problems in FreeCAD FEM. It’s sometimes even jumping to the side so much that all labels are covered.

luzpaz, 2025-08-14:
cc @FEA-eng still relevant ?

FEA-eng, 2025-08-16:
> cc [@FEA-eng](https://github.com/FEA-eng) still relevant ?

Yes, it's still a major flaw of the color bar.

[image]