FreeCAD/FreeCAD #11666: [Problem] Geometry selection for FEM features is unintuitive and inconsistent

FEA-eng, 2023-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Geometry selection for FEM features definitely needs some improvements. Let's take [FEM ConstraintForce](https://wiki.freecad.org/FEM_ConstraintForce) as an example:
- GUI says "Select multiple face(s), click Add or Remove" - this is the first problem because the order suggested by this message is incorrect - one has to click the Add button first and then select a geometrical entity
- to make a selection for Direction, one has to click the Add button again to deactivate it, then select a face/edge and press the Direction button
- 
Now let's take a look at [FEM ConstraintContact](https://wiki.freecad.org/FEM_ConstraintContact):
- GUI says "Select master face, click Add or Remove" for the first selection and the suggested order is correct this time - if you click the Add button first, it shows an error message ("Nothing selected!")
- to fill the second selection field, there's no need to deactivate the first Add button, one just has to select another face and click the second Add button (likely because only one face can be selected for each field, unlike in the case of FEM ConstraintForce)

So the first thing to do would be to correct the GUI tips for features like FEM ConstraintForce so that they don't suggest an incorrect selection order. But IMO it's also necessary to unify the selection for all features and make it as user-friendly as possible.

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-03-24:
It's not even consistent in FEM. Some tools require a selection before you click on the tool, otherwise the user gets an error. Other tools do not work with a pre-selection and you have to select all again with the tool active. In another tool you always have to click the *select* button between each new selection.
It should be:
1. all tools in FEM should use the pre-selection as input where possible.
2. with an open task dialog and option to select elements it should be like the Fillet tool in PartDesign WB. A mode to just select and deselect, of if possible not even a special mode. Just see all selected andeither click on them to deselect or select some elements which are not selected.

FEA-eng, 2024-03-24:
@maxwxyz Right, FEM suffers from extreme inconsistency in this regard. But geometry selection in this workbench is one of the biggest problems in general - you can't select interior entities or multiple entities at once (with a selection box), often making selection extremely tedious. But we have separate issues for that so let's go back to this topic. IMO an ideal approach would be what is done in many other FEA preprocessors - you just start a tool and select geometry, holding SHIFT to add and CTRL to remove items. Of course, they are highlighted but there's no visible list like in FreeCAD. I know that it's not really the FreeCAD way though so I don't mind a list but it should be way more user-friendly and consistent, as you described.

maxwxyz, 2024-05-18:
This problem is also in the rigid body constraint task dialog.

FEA-eng, 2024-05-18:
> This problem is also in the rigid body constraint task dialog.

It uses the same selection panel as the displacement or fixed constraint and they should share the same rules in this regard. But in the future, selection should be unified and simplified for other tools. It's the worst for contact/tie, force and such. Boundary conditions (the aforementioned ones and others) are not that bad.

HoWilgh, 2026-10-23:
Would this macro help?
https://github.com/HoWilgh/FreeCAD-macros/tree/master/FEM/SelectionSet

```
# FreeCAD macro: SelectionSet

Volume-based selection of faces or solids in FreeCAD. SelectionSet: define a “selection shape” (e.g. a box, sphere, or PartDesign body), then store which faces or solids of another object lie inside or intersect it as a SelectionSet in the model tree. SelectionSetLink: combine one or more SelectionSets (and/or direct shapes), add and subtract as needed, then apply the result to a FEM constraint, material, or any object with a References property—so you drive FEM geometry from your selections in one place.
```
...works also with coordinates.

FEA-eng, 2026-10-25:
> Would this macro help?

I addressed this proposal on the forum, but in short words, the main problems with selection in FEM are that it's hard to select multiple objects and that the selection UI is counterintuitive. However, the named selection system with bounding box approach could make a lot of sense too.