FreeCAD/FreeCAD #11671: [Feature Request] TechDraw: Missing Features / Improvements

maxwxyz, 2023-12-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@WandererFan I created this issue to collect missing features / icons / lines or improvement based on current standards which I've stumbled upon in the TechDraw WB. If needed, for each implementation a new issue could be created.

- [ ] Workpiece edge symbols (**ISO 13715**) [image]
- [ ] Line styles / general representations of drawings (**ISO 128-24** / **ISO 128-20**)
- [ ] Section cuts / representation of section lines / arrows (**ISO 128-40**) [image]
- [ ] Hatching / default representation of materials (**ISO 128-50**) @davesrocketshop maybe interesting for your implementation
- [ ] General page layout (**ISO 5457**) and title block template (**ISO 7200**)
- [ ] #11112 General support for ISO GPS (GD&T / ISO 8015 ), see **ISO 14638** for the matrix model reference of all GPS standards // [Wikipedia](https://en.wikipedia.org/wiki/Geometric_dimensioning_and_tolerancing#Documents_and_standards)
- [ ] Symbolic presentation and indication of adhesive, fold and pressed joints (**ISO 15785**)

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-08:
@davesrocketshop how is it currently implemented? The hatching seems only relevant for drawings / all 2D representations in TechDraw, Draft, Arch/BIM and maybe in the future for the closed clipping plane profiles (view section cuts), therefore a simple 2D rectangle displaying the hatch pattern next to the 3D sphere representation would be sufficient in my opinion.

Currently hatching is implemented in TechDraw with presets (.svg files) https://wiki.freecad.org/TechDraw_Hatching
It would be great if the material editor could have in the appearance tab settings to adjust the hatching of the individual material, e.g. change line spacing, angle, thickness,...
https://forum.freecad.org/viewtopic.php?style=1&p=717232#p717232

Here is a excerpt from ISO 128-50
[image]

davesrocketshop, 2023-12-08:
I've currently implemented to models. For simple cross hatches as defined in FCPAT.pat, here is an example material card:

```
---
# File created by FreeCAD 0.22.0 Revision: 35279 +19 (Git)
General:
  UUID: "a9146fcf-4168-4065-8a58-ec78735e471d"
  Name: "Diamond4"
  Author: "David Carter"
  License: "All rights reserved"
AppearanceModels:
  PAT:
    UUID: "0326c759-4e3d-46ca-bb7d-146ebebea65e"
    Hatch Color: "(0,0,0,1)"
    Hatch Pattern: |2
      *Diamond4, 45 diagonals L & R, Solid, 4.0 mm separation
      45,0,0,0,4.0
      -45,0,0,0,4.0
    Hatch Scale: "1"
    Hatch Weight: "1"

```

For file based patterns (SVG):

```
---
# File created by FreeCAD 0.22.0 Revision: 35279 +21 (Git)
General:
  UUID: "55f2480e-53bb-4b99-ba8b-5d0420929bf2"
  Name: "Example"
  Author: "David Carter"
  License: "All rights reserved"
AppearanceModels:
  Pattern File:
    UUID: "c6596294-e97d-4812-87db-28e1d66521a3"
    Pattern Color: "(0,0,0,1)"
    Pattern File: "C:/Users/dcarter/Documents/Projects/freecad-source/src/Mod/TechDraw/Patterns/general_steel.svg"
    Pattern Scale: "1"

```

The latter could be easily improved with the addition of an SVG type that would allow it to be embedded. The current Image type doesn't handle SVGs.

If all that's required is to view the cross hatch, then that is an easy addition.

maxwxyz, 2023-12-08:
For a first step displaying the matching hatch pattern next to the 3D sphere would be sufficient. 
If possible even the option to change it, maybe if easier firstly be allowing the user select a pattern (like your second option), the ideal option would be to change the pattern parameters itself in the material editor, like it is possible to adjust the 3D representation in the editor. For easy line pattern that would be trivial I guess, looking at your first option:

> Hatch Color: "(0,0,0,1)"
    Hatch Pattern: |2
      *Diamond4, 45 diagonals L & R, Solid, 4.0 mm separation
      45,0,0,0,4.0
      -45,0,0,0,4.0
    Hatch Scale: "1"
    Hatch Weight: "1"

I think its difficult if other geometries are needed for the hatch (repeated groups of lines, broken/dashed lines, circles, crossed lines, dotted lines...)

davesrocketshop, 2023-12-08:
The model definitions for these will be part of my next merge, as will the current PAT and file definitions. I'll try to get the preview in there but that may take a little bit longer due to competing priorities. Similar if embedded SVG files are desired and I think they are.

It should also be possible to add a simple importer for .pat files, but again competing priorities.

davesrocketshop, 2023-12-08:
> For a first step displaying the matching hatch pattern next to the 3D sphere would be sufficient. If possible even the option to change it, maybe if easier firstly be allowing the user select a [...]

I don't like the pattern next to the pattern is desirable as it will detract from the preview. A separate preview tab at the top seems a better idea. Editing the patterns is a matter of knowing the syntax, as would be updating the display. It's now in the queue

davesrocketshop, 2023-12-08:
> I did not mean that the user would go in and edit the syntax, this should be done with normal value fields / sliders or with graphical representation, like currently the 3D appearance.

In the thinking queue

WandererFan, 2024-02-19:
>>> Line styles / general representations of drawings (ISO 128-24 / ISO 128-20)

This one is fixed, no?

WandererFan, 2024-02-19:
>>> Section cuts / representation of section lines / arrows (ISO 128-40) 

What is wrong with section lines & arrows?  The picture is showing ANSI/ASME section line, but the item refers to ISO128.

The section cuts issue is about different cut face hatching for different source objects?

davesrocketshop, 2024-02-19:
> > > > Line styles / general representations of drawings (ISO 128-24 / ISO 128-20)
> 
> This one is fixed, no?

It is supported in the materials, but it's not currently used anywhere. 'Fixed' would be the wrong word.

maxwxyz, 2024-02-19:
> > > > > Line styles / general representations of drawings (ISO 128-24 / ISO 128-20)
> > 
> > 
> > This one is fixed, no?
> 
> It is supported in the materials, but it's not currently used anywhere. 'Fixed' would be the wrong word.

The new line styles were implemented in TD preferences. I've updated the original post.

maxwxyz, 2024-02-19:
> > > > Section cuts / representation of section lines / arrows (ISO 128-40)
> 
> What is wrong with section lines & arrows? The picture does not look like v0.22dev.
> 
> The section cuts issue is about different cut face hatching for different source objects?

The idea was to remove the dotted/dashed line (marked with arrows) and only leave the edges, where the complex section cut makes a turn (as in the original post with a reference image from the ISO)
[image]

luzpaz, 2024-04-15:
Do we want to split this ticket up ?

davesrocketshop, 2024-04-15:
> Do we want to split this ticket up ?

I think so. Adding it to the materials (done), material preview (not done), and using it Techdraw are different issues. The first 2 can be remain grouped together, while the last is better handled by the TD folks.

maxwxyz, 2024-04-15:
I will split it tomorrow.

maxwxyz, 2024-04-21:
I've created individual issues so I will close this here. They are linked above.

closed 2024-04-21 by maxwxyz