FreeCAD/FreeCAD #11674: [Problem] Nonsensical errors when moving unconstrained shape in sketch

JMaravalhasSilva, 2023-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When drawing a sketch with circles (with defined diameters) centered on the vertices of a reference hexagon and defining the diameter of its circumscribed circle, FreeCad reports multiple errors in quick succession when attempting to move the shape around in the sketch, with the shape making some bizarre movements.

Even though the sketch has 3 unconstrained DoFs (X and Y movement, and rotation over itself), this is clearly not intended behavior since the shape being moved is itself properly defined.

[image]

When attempting to move the shape around:
[image]

Notice how the errors are about some radius which should be positive - something that should never happen since all radii are clearly well defined.

[Screencast from 07-12-2023 03:59:28.webm](https://github.com/FreeCAD/FreeCAD/assets/24859764/8be6ee40-7fab-4cd5-b5b8-47c283ac4153)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

To replicate, download this minimal example and move the shape around:
[problem_report.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13594712/problem_report.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-07:
Confirmed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods: 
```

FEA-eng, 2026-04-03:
Still relevant

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20260401 (Git shallow)
Build date: 2026/04/01 04:43:57
Build type: Release
Branch: (HEAD detached at 33dcd31f5)
Hash: 33dcd31f5e2175e7dcdfd3caf1d9d86440919bfc
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FEA-eng, 2026-10-13:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261006 (Git shallow)
Build date: 2026/10/06 21:00:40
Build type: Release
Branch: (HEAD detached at 3424c4106)
Hash: 3424c410627263394f701aeec5b61d7bab7bb3a1
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```