FreeCAD/FreeCAD #11676: [Problem] Tree: Drag&Drop hectic. Many unexpected behaviors.

PaddleStroke, 2023-12-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Origins should not accept drag/drop of their elements. (unless I'm missing some specific workflow?)

- Issue 1 :
Trying to drop something on the origin object gives an error and the object is not dropped, which is the correct action, though the error message seems wrong:

> Bad object "Unnamed#Sphere"(Part::Sphere) linked to the Origin of "Unnamed#Part004"

However if you try to drop something inside the origin, then behavior gets crazy. See below the thorus took the place of the Z-axis004 that got kicked out.

[image]

Issue 2 : 
If you try to drop the origin object in a group for example, then it duplicate and a copy seems to get in the group.

[image]

The drag and drop behavior is hectic and weird and needs to be fixed. I could creates a lot of issues with all the different weird things that are happening. And that's not even mentioning the Ctrl and Alt modifiers.

From this video : https://www.youtube.com/watch?v=f1raOT4thEM&ab_channel=LeiZheng I seem to understand that this behavior is actually a feature. But it is not behaving correctly.

The default behavior from the drag&drop should be the normal behavior : either the object moves if it can, or does not move at all. But it should not replace or create a copy or create a link etc randomly.

Special functions like 'replace' 'copy' or 'create link' should be achieved only with modifier keys.

@realthunder can you please advise how to fix this situation?

Related discussion : https://forum.freecad.org/viewtopic.php?p=723942#p723942

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35306 +17 (Git)
Build type: Release
Branch: asm_card9
Hash: a601f402437217a64cccd1b5084b6a7738837ffe
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk , OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60k
  * Assembly3 0.12.0
  * Assembly4 0.50.2
  * Curves 0.6.8
  * ExtremeProLight 1.5.0
  * FreeCAD - Shortcut.lnk
  * FreeCAD_FastenersWB-master 0.4.56
  * Ondsel 2023.10.31.01
  * OpenDark 2023.10.6
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
```
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2023-12-21:
@wwmayer @chennes Any comment on this drag & drop behavior?

If no one is against it I propose to : 
- Remove special modes from occuring when basic drag&drop to avoid bugs and confusion.
- Special modes only when modifiers keys are pressed.

realthunder, 2023-12-22:
The problem is that there is too many special modes, which makes it impractical to assign an one to one relation to some modifier key combination. The current behavior is for the tree to query capability of the dragging and dropping objects and their parent before executing the action. The modifier key is used as a hint for preference in case there are more than one possible actions. I agree we can improve the modifier hint part of the logic. But In the end, it is the object that decides what to do in the final action. For a group, the drag and drop may be interpreted as adding object as membership. For an uninitialized Link, it maybe link assignment. Some parent allows to drag out object, some may not. Some container only accepts drop if the object can be dragged out of its current parent, etc. Basically, the object can do whatever it wants with the action. I don't see how it is possible to control it at Tree level. Just like the origin group problem mentioned, it can only be fixed in the object view provider's code, not at the Tree level.

PaddleStroke, 2023-12-22:
> The problem is that there is too many special modes, which makes it impractical to assign an one to one relation to some modifier key combination. The current behavior is for the tree to query [...]

I still think the current situation induce a lot of confusion. Even with tree.cpp right in front of my eyes I couldn't fully understand what should happen when. And then there are the weird situations I mentioned before.

What about : 
- Drag drop only try to move by default.
- When Ctrl is pressed : dropping opens a small dialog with the possible actions to choose from

> But In the end, it is the object that decides what to do in the final action. For a group, the drag and drop may be interpreted as adding object as membership. For an uninitialized Link, it maybe [...]

Well I can't say this is true because I couldn't achieve the behavior I wanted : https://forum.freecad.org/viewtopic.php?p=723942#p723942
Can you please advise what I am missing to do : 

> - JointGroup : User should not be able to drag it / drop it anywhere.
> - Nothing should be able to be put inside JointGroup by drag/drop
> - Nothing should be able to be removed from jointGroup by drag/drop.
Thanks

realthunder, 2023-12-22:
The behavior in the video is implemented in `ViewProviderDocumentObject::replaceObject()`. And I agree this action should only be allowed with explicit request. You can try disable it by comment out the whole condition block [here](https://github.com/FreeCAD/FreeCAD/blob/f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb/src/Gui/Tree.cpp#L1750), and see if there is any other undesired behavior.

> * Drag drop only try to move by default.

I agree we should refine the condition for choose the default action, but allowing only move for default is too restrictive in my opinion.

> * When Ctrl is pressed : dropping opens a small dialog with the possible actions to choose from

Yes, I wanted to implement that too (not exactly on Ctrl, but same concept). But haven't got time to do it yet.

PaddleStroke, 2024-01-08:
Thanks @realthunder 
I made a PR to comment out this block as you said and it does fix the most important bug (torus replacing origin elements).
Other behaviors, such as creating a copy on drop are still there. I still think this is confusing. At least a qmessage should inform the user of what is going on : 
"Cannot drop here, create a copy? Yes No Yes always"

chennes, 2024-01-13:
Let's ask the Design Working Group to have a look at this and come up with some consistent guidelines for developers. @obelisk79, can you all discuss the handling of drag and drop in the tree?

obelisk79, 2024-01-15:
We are working on a set of draft guidelines for behavior. I should have a solid draft for submission to the design guide in a few days.

maxwxyz, 2024-02-25:
related issue: https://github.com/FreeCAD/FreeCAD/issues/12055

maxwxyz, 2024-02-26:
Comments from the complaint session: 
It's currently not possible to drag something from one object to another. You have to first drag it to the root and then you can drag it to the other object, because you cannot drag it from an object directly to the other like a sketch.

krushia, 2024-06-15:
> We are working on a set of draft guidelines for behavior. I should have a solid draft for submission to the design guide in a few days.

Did this ever happen? I'm starting to develop my own workbench and very confused about the "proper" way to do things. See related bug https://github.com/FreeCAD/FreeCAD/issues/14146

closed 2024-06-17 by PaddleStroke