FreeCAD/FreeCAD #11684: Can't link Leaderline and surface finish symbol

danergo, 2023-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm using latest FreeCAD, and wish to add surface finish symbols to my design.

According to documentation here: https://wiki.freecad.org/Release_notes_0.21#TechDraw_Workbench

> [LeaderLine](https://wiki.freecad.org/TechDraw_LeaderLine) tool can be used to properly refer oriented symbols to the edges of an object.

But I can't link leaderline and surfacefinishsymbol. I wish the symbol to move together with its referenced view.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/World (en_001)
Installed mods: 
  * fasteners 0.4.71
  * lattice2 1.0.0
[/code]
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-14:
@pavltom do you have any availability to take a look at this ?

pavltom, 2023-12-16:
Hi @luzpaz, I have done some basic research, it should be quite feasible, but with all the Christmas preparations, the time is scarce. If no one decides to implement it earlier, I would try it by the beginning of January.

pavltom, 2024-02-12:
Hi @luzpaz, my aim was to resolve this issue by delivering the PR above and from my point of view it is resolved. However if anybody (@danergo?) could verify and confirm the resolution, it would be nice.

luzpaz, 2024-04-04:
@danergo please follow-up. Closing this ticket in the meantime. Please re-open if necessary. 

Fixed by #12342

closed 2024-04-04 by luzpaz