FreeCAD/FreeCAD #11687: [Problem] TechDraw Task View not opening / not ideal for dark themes

maxwxyz, 2023-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@WandererFan  most TechDraw features do not automatically open the new task dialog in overlay mode (transparent and auto show) see also: #11637  

Some icons are hardcoded into the dialog that they cannot be themed or are hard to see in dark themes, here the weld icon:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git)
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.8
  * CurvedShapes 1.0.5
  * Curves 0.6.15
  * Defeaturing 1.2.1
  * fasteners 0.4.71
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.11.27
  * sheetmetal 0.3.12
  * woodworking 0.21.30486
```
```

### Subproject(s) affected?

Techdraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-12-09:
I will need some help on this one.  I don't even know what overlay mode is.  Who is the expert?

maxwxyz, 2023-12-09:
I could help you. The dialog for weld symbols is working great in dark mode!

WandererFan, 2023-12-09:
Surface finish is in Python, weld symbols are in c++, so we can't just compare them to see what's different (I think.  My Python Gui skills are pretty weak.).  The other Python Guis in TD are AddOffsetVertex, MoveView and ShareView.  Do any of them behave nicely with overview mode?   Can you come up with another TD task that misbehaves?

@edi271 you are the wiz on surface finish.  Any comments?

maxwxyz, 2023-12-09:
MoveView and ShareView are similar and are as all dialogs ok in dark mode. The only dialog is the weld symbol, as the images seem to have a white background. If the symbols inside the task dialog could be a grey, I think they can be themed with the underlying button to be readable in dark mode (and the default theme), what is your take on this @obelisk79 ?

In general, the task views do not show in TechDraw automatically (when in docked auto show mode): #11637 
I'm not sure if this is something TechDraw / the command has to fix or if this is a general problem. Most other features in other WB work normal as expected. There is a demo video in the issue above.

obelisk79, 2023-12-09:
The symbols don't follow any art guidelines as far as I can tell. They should be redrawn in accordance with the current art guidelines so the symbols will look acceptable on any background. Dark or light

edi271, 2023-12-11:
I think the issue is independent of C++ or Python.

Creating the 6 push buttons the **Surface Finish Symbols** tool provides the following steps:

1. In an ui file by using Qt-Designer push buttons having an empty text label are created.
2. The tool creates small svg files showing the symbols.
3. These svg files are added as icons to the buttons.

Conclusion: The push buttons show icons. Icons do not change if the theme is changed, see the tool icon in the head line. These icons cannot become designed like any icon, because they show symbols which appearance is fixed in the ISO standard. There have to be black lines on white background. The issue is, that the buttons (created by Qt-Designer) change their background if the theme is changed. Maybe there is an option avoiding the change of background of buttons it the theme changes. To invert the color, like the OK button does is no solution, because "OK" is a string, not an icon.

To the below picture showing the symbol:

A QGraphicsView is created using Qt-Designer. This view contains a QGraphicsScene which holds several buttons. The issue is too, that the buttons (created by Qt-tools) change their background if the theme is changed.

luzpaz, 2023-12-24:
@edi271 your wrote:

> To the below picture showing the symbol:

but there is no image.

WandererFan, 2024-02-18:
There is a mountain of rules regarding these symbols.  

I don't know if it is against the rules, but I've never seen an example of welding symbols that was anything other than dark on light.  Would a colour break at the edge of the icon help?

ISO Surface Roughness Symbols Terminology
Engineers Edge:
https://www.engineersedge.com/manufacturing/_iso_surface_roughness_symbols_terminology_13090.htm

Welding Symbols
https://www.weldinginfo.org/welding-technology/welding-symbols-types-diagrams-dimensions-advantages/

maxwxyz, 2024-02-18:
Yes, they are black on white on the drawing page but it just looks off in the task dialog with a dark theme.
@obelisk79 do you have an idea how to better display them in the dialog? Just black icon files with a transparent background and a lighter button?

obelisk79, 2024-02-18:
That would be one approach. Obviously this can get a bit tricky, but I think an elegant solution would be to use an SVG for the icons which checks the luminance of the background color and modifies the fill for the SVG's in the task panel (not the drawing itself) and switches between white or black for the symbols to maintain contrast.

I did a lazy google image search and was able to find some examples of these symbols with a light on dark effect. I am less familiar with the iso standard for this type of drawing, but it would stand to reason that the colors used are less important than readability and the symbols themselves.

https://primeweld.com/blogs/news/welders-guide-to-welding-symbols

maxwxyz, 2024-03-03:
To add more info here:
drop down menus sometimes overlap, no padding to the icon itself, lines do not meet in the button icons:
[image]

luzpaz, 2026-09-22:
> Well, not relevant to me since I don't use dark theme and I don't know what overlay mode is. YMMV.

lol