FreeCAD/FreeCAD #11689: [Problem] Assigning a random color cannot be undone

prokoudine, 2023-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create a part/body. Select, right-click, "Random color". This change isn't registered as an undoable action and cannot be undone. The project isn't marked as dirty either.

### Full version info

```shell
OS: Ubuntu 23.10 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35138 (Git)
Build type: Unknown
Branch: main
Hash: 0a094d487249dab38b279120599de4b592c98e83
Python 3.11.6, Qt 5.15.10, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * ExtremeProDark 2.9.0
  * DynamicData 2.53.0
  * Assembly4 0.50.4
  * ExplodedAssembly
  * Ondsel-Lens.backup1696028800.7281785 2023.8.10.02 (Disabled)
  * sheetmetal 0.3.12
  * BIM 2021.12.0
  * A2plus 0.4.60n
  * Render 2023.8.18
  * Ondsel-Lens.backup1698677147.1083882 2023.10.26.01 (Disabled)
  * Rocket 3.2.3
  * Assembly3 0.12.0
  * boltsfc 2022.11.5
  * OpenDark 2023.10.30
  * freecad-mbdyn-dynamics-workbench
  * Ondsel-Lens 2023.10.31.01
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-08:
There is a related [fine-tuning](https://wiki.freecad.org/Fine-tuning) parameter: `PropertyView/AutoTransactionView`. But that parameter only works for 'manual' changes to view properties.

luzpaz, 2023-12-14:
I imagine the issue is somewhere in this function?   
https://github.com/FreeCAD/FreeCAD/blob/abd411f934e56469e4748b5ecf688eeba0108f2d/src/Gui/CommandFeat.cpp#L68C1-L75

PR #12187 (2024-01-30): Make the random color assignment undo-able
Create a transaction in the currently selected document to allow the user to roll-back the color assignment.

Closes #11689
commit 044d90b4 (2024-01-30): Make the random color assignment undo-able

Create a transaction in the currently selected document to allow the
user to roll-back the color assignment.

Closes #11689
closed 2024-01-30 by chennes