FreeCAD/FreeCAD #11690: [Problem] Version 0.21.1 Sketcher Angle Constraint

bigdipperaviation, 2023-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When system units are set on "Imperial for Civil Engineering"; angle constraints behave oddly. When you set an angle constraint, the angle is graphically displayed on the angle between two lines; however, the value in the data entry box is "0.0000". See attached screenshot. When units are switched to "Imperial Decimal", there is not issue. The angle constraint does not behave this way in the copy version 0.20.1 that I have. 

Version 0.21.1 Not Functioning
[image]

Verision 20.1 Functioning Correctly
[image]

Forgive me if there is something simple code-wise that I could correct to resolve this issue. I am not nearly as proficient as some of you experts on here.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.60n
  * ExplodedAssembly
  * fasteners 0.4.71
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-09:
It is present on Linux too.

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35327 (Git)
Build type: Debug
Branch: main
Hash: 32a0e0c6ae7fdd6e179065c247d53b8497bf1db2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

adrianinsaval, 2023-12-11:
possibly related to the changes in how these input fields are parsed, @ConnorWorrell might be able to shed some light here

ConnorWorrell, 2023-12-12:
Off the cuff I would guess its related to this https://github.com/FreeCAD/FreeCAD/commit/e85c383bff4f4628edc68daec80156f4be3307f3

Looks like the way this Schema displays angles got changed.

https://github.com/FreeCAD/FreeCAD/blame/0649e38bffc9fef46f46123f53dc0790db001fdc/src/Base/UnitsSchemaImperial1.cpp#L363-L366

@Syres916 let me know if this is where the bisect lands, if it is then it should be trivial to add this format to the expression parser.

Syres916, 2023-12-12:
@ConnorWorrell I've had to abandon this, all I'm doing is fighting uncompilable commits and over 12 hours of my time is the limit. The best I can say is the last good commit I have is `68e337670e227889217652ddac593c93b5e8dc94` and the first bad commit is `270bb80e797282318111453f0d32ce94e383cdf2`

ConnorWorrell, 2023-12-14:
@Syres916 Thanks for getting the head start on the bisect.

Finishing the bisect landed at https://github.com/FreeCAD/FreeCAD/pull/8982, which was a QuantitySpinBox update.

Looks like this unit format just needs to be added to the Expression Parser and all will be well again. I'll take a swing at it this weekend, should be trivial, the process is almost identical to https://github.com/FreeCAD/FreeCAD/pull/9397.

ConnorWorrell, 2023-12-18:
@bigdipperaviation if you are able to, can you try out this PR, it should fix this issue: https://github.com/FreeCAD/FreeCAD/pull/11757

FEA-eng, 2025-08-08:
Still relevant