FreeCAD/FreeCAD #11695: [Problem] Navigation style using mouse button 4 & 5 (or other, possibly custom?)

fischer-felix, 2023-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since I have a mouse with two side buttons (normally forward/back) I would like to be able to use these in my navigation.

More concretely, I am using the Blender style. In Blender, I am able to map Pan / Zoom (normally Shift + Mouse2 and Ctrl + Mouse2) to Mouse 4 and 5.

I would like to achieve the same thing in FreeCAD. An added bonus would be being able to configure what mouse button does what, but I have already seen that this seems to be on the roadmap.

I would also be open to playing around with the source code, if give some pointers about the general way navigation styles work.

### Full version info

```shell
[code]
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * ThreadProfile 1.84.0
  * sheetmetal 0.2.63
  * Launcher
  * InventorLoader 1.3.0
  * Render 2022.3.0
  * BOLTSFC 2022.11.5
  * fasteners 0.4.25
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-12-10:
> An added bonus would be being able to configure what mouse button does what, but I have already seen that this seems to be on the roadmap.

That would indeed be perfect. There are some navigation styles that I'd like to have in FreeCAD.

> I would also be open to playing around with the source code, if give some pointers about the general way navigation styles work.

Check this wiki page: [Adding a new mouse navigation option to FreeCAD](https://wiki.freecad.org/Adding_a_new_mouse_navigation_option_to_FreeCAD)

fischer-felix, 2023-12-14:
Definitely going to look into it some more.
I don't really have any experience with cpp in terms of UI dev and QT, but maybe a first step could be adding a new variable to FreeCADs `user.cfg` file.
That way there could be a "custom" style, that just gets it's bindings from that file. Not as polished, but possibly easier to implement?

Do you by chance know how these parameters are read from the file?

luzpaz, 2023-12-24:
For reading parameters, check out https://github.com/FreeCAD/FreeCAD/issues/11632#issuecomment-1840161829

maxwxyz, 2024-02-25:
@Rexbas ping :)

Rexbas, 2024-02-25:
We could easily add two parameters that describe what mouse button 4 and 5 should do. We just have to handle them in all individual existing navigation styles. @fischer-felix, have you done some work on this already or should I try to implement it?

I would stay away from a custom style for now. Ideally we create a `CustomNavigationStyle` that reads bindings from a file and then simply works. Then we could rewrite existing styles to be presets of key bindings and this would probably make maintaining navigation styles easier. I have been thinking about this in the past and might attempt to implement this, maybe this year.

fischer-felix, 2024-02-25:
I have played around with it a bit and could get it to work using a hard-coded navigation style (it was as simple as replacing e.g. the number 2 with 4 in a couple of places).
I am, however, not familiar enough with cpp and haven't got the time right now to implement it reading values from a file.

I think what you are suggesting would be a great step in the right direction to make FreeCAD more customisable.

luzpaz, 2024-03-18:
@fischer-felix can you link to a branch that other can look at to examine the code ?

fischer-felix, 2024-03-18:
> @fischer-felix can you link to a branch that other can look at to examine the code ?

Sorry that it took me a while, finally found some time to get around to it.
The code can be found here https://github.com/fischer-felix/FreeCAD/tree/add_mb.
I hope it still works, I pulled unrelated commits from upstream, so not entirely sure.