FreeCAD/FreeCAD #11698: [Problem] Some labels use hardcoded colors making them illegible

Beep6581, 2023-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Some labels seem to use hard-coded colors, making them difficult or impossible to see depending on the color scheme (style sheet).

[image]

This "DoF(s)" label is very difficult to see in most dark stylesheets. The example screenshot uses the "Behave-dark" style sheet.

The color should not be hard-coded but theme dependent.

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.71
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-10:
This has been fixed in the 0.22dev builds https://github.com/FreeCAD/FreeCAD/commit/c6eb321f5403c8b50818d766f263515ea0e7a449 so you can manually overwrite your copy in 0.21.2

Beep6581, 2023-12-10:
I assumed it was hard-coded in the Qt source code based on the familiar blue color, but that commit shows it can just be styled - good. However, it seems the fix applies only to the `Behave-dark` style?

On a related note, does Qt have something similar to Gtk's [Inspect](https://wiki.gnome.org/Projects/GTK/Inspector) tool?

Beep6581, 2023-12-10:
btw. fix confirmed for `Behave-dark` style.

adrianinsaval, 2023-12-16:
@MisterMakerNL is this still a problem in any of our other current stylesheets?

luzpaz, 2023-12-24:
Thanks for the heads up beep6581

bump @MisterMakerNL

MisterMakerNL, 2023-12-24:
It's been changed to a less full blue, although it is still blue. It's better readable
[image]

closed 2023-12-24 by adrianinsaval