FreeCAD/FreeCAD #11700: [Problem] Transparent menu background when using Light/Dark/Darker style sheets

Beep6581, 2023-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Menu items have a transparent background when using any of the Light/Dark/Darker style sheets, making it difficult to see the menu contents.

[image]
[image]

Behave-dark, ProDark and No Style Sheet do not exhibit this problem.

[image]

Manjaro/KDE.

### Full version info

```shell
[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.71
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-12-11:
@MisterMakerNL any clue?

MisterMakerNL, 2023-12-12:
This was already reporter and fixed. Although I would think it should be in already fixed in 21.2. So that is a bit weird.

adrianinsaval, 2023-12-13:
Can you link to the fix?

maxwxyz, 2024-02-27:
is this still relevant?

MisterMakerNL, 2024-02-27:
Since I have not heard any others complaining it should have been fixed here https://github.com/MisterMakerNL/FreeCAD_MisterMaker/commit/125406a7641452c3882b6eeddb09d1efd74b4185

maxwxyz, 2024-02-27:
@MisterMakerNL was this merged in main?

closed 2024-02-27 by maxwxyz