FreeCAD/FreeCAD #11712: FreeCAD 0.21.1 Euler angle blocked between -90° and 90° on part and body.

Almisuifre, 2023-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since the last release, it is no longer possible to rotate 360° on the X axis when using placement.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * fasteners 0.4.68
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

Fair:
- Place a body in part design mode
- Select investment
- Select euler angle
- Select y and type 180. Everything stops at 80. If done manually with the arrows on the right, it goes from -90° to 90° only.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-13:
[This commit](https://github.com/FreeCAD/FreeCAD/commit/e171a1f6e175bbff4cdcbfd5309143488f7ccffc) introduces it.

@Almisuifre, isn't pitch supposed to be in this range?

Almisuifre, 2023-12-13:
I’m not used to testing. Is the committee planned to be introduced in 0.21.2?

xtemp09, 2023-12-14:
I don't think I understand your reply correctly. The changes were intentionally made long time ago. If you look here, page 3,[^1] seems like α, γ ∈ (−π, π] and β ∈ [−π/2, π/2)

> Given R ∈ SO(3), we wish to find angles α, γ ∈ (−π, π] and β ∈ [−π/2, π/2) that satisfy

[^1]: https://bjpcjp.github.io/pdfs/robotics/MR_apndxB_rotation_representations.pdf

Almisuifre, 2023-12-14:
But it’s not practical at all. Especially why the other two axes are not subject to this rule. Wouldn’t there be a way to unlock this situation with a check box for example?

xtemp09, 2023-12-14:
Perhaps, you might want to start a discussion at FreeCAD forum.[^1]

[^1]: https://forum.freecad.org

Almisuifre, 2023-12-14:
Negatgif, it’s the same. On the contrary, I think it’s worth seeing if it’s permanently editable, and talking about it in the forum if necessary. Not the other way around!

luzpaz, 2023-12-14:
Please link to the discussion when you create it

luzpaz, 2023-12-23:
What's the proposal here for a fix ?

xtemp09, 2023-12-23:
Might ven. @wwmayer have something to say here?

wwmayer, 2024-03-14:
The changes were needed to avoid ambiguities when converting Euler angles to a quaternion and back.
Here the relevant forum thread: https://forum.freecad.org/viewtopic.php?p=640998#p640998

Almisuifre, 2024-03-14:
Good news.

I just looked at version 0.21.2. Same persistent problem.

I would be happy if each axis could be -180° to +180° with a center at 0° versus -90° +90°.

FEA-eng, 2025-06-08:
Still relevant

Almisuifre, 2025-06-09:
That’s right, the problem still exists. The only workaround is to put the created object in a Part.

The 1.0.0 and 1.0.1 has the same behavior.

luzpaz, 2026-04-10:
@pyro9 any interest in looking into this ?

FEA-eng, 2026-10-11:
Should still be relevant