FreeCAD/FreeCAD #11714: [Path Workbench - Custom command modifies values for A and B parameters ]

dasnorkel, 2023-12-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As described in https://forum.freecad.org/viewtopic.php?p=715182#p715182.

The custom command "G0 A0 B180" is entered and "G0 A180 B0" is output to file.

I believe that somewhere in App/Command.cpp a call gets made to getYawPitchRoll
which modifies the input to standard Euler representation.
Since I wanted to use the "custom" command for a Pocket NC, which can rotate A between -25 to 135 degrees and B between -9999 and 9999 degrees, I have to manually update the outputted gcode files.

Note: I used FreeCAD as it is (i.e. only x,y,z) but preambles the rotation setup for the PNC to the correct plane.

### Full version info

```shell
[code]
OS: NixOS 22.11 (Raccoon) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.1.Unknown
Build type: Release
Python 3.10.12, Qt 5.15.10, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * ExplodedAssembly
  * Curves 0.6.9
  * freecad.gears 1.0.0
  * FeedsAndSpeeds 0.5.0
  * Assembly4 0.50.2
  * InventorLoader 1.4.0 (Disabled)
  * LCInterlocking
  * sheetmetal 0.3.0
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2025-07-06:
This seems to be happening farther downstream.  
It does not occur in the linuxcnc post or any of the refactored posts that I've tested.

If you can use one of the refactored posts, let us know and we'll close this issue since it will eventually be resolved by the refactoring.  If you need one of the legacy posts, let us know and we'll see if we can find the root cause in the post.

FEA-eng, 2026-01-13:
> I am not really sure what to test on. FreeCAD-daily repo ?

Possibly with the newest weekly build from here: https://github.com/FreeCAD/FreeCAD/releases

There are AppImage builds for Linux.

sliptonic, 2026-10-14:
If I understand correctly,  Root cause: The error is in Command::transform() (src/Mod/CAM/App/Command.cpp:327). It converts A/B/C to a rotation (yaw-pitch-roll) and back. This changes the values even with an identity placement:

However, downstream handling is correcting things.  I've retested and can't duplicate the problem any more.  
While the defect remains, I'm leaving it alone.  The command structure is likely to be touched when we work to remove the tool orientation assumptions.  We can address then.  

Closing this now.  Will reopen if someone shows a failing case.

closed 2026-10-14 by sliptonic