FreeCAD/FreeCAD #11719: [Problem] FEM Stress linearization plot can't use the stress tensor components

FEA-eng, 2023-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, a major limitation of [FEM_PostFilterLinearizedStresses](https://wiki.freecad.org/FEM_PostFilterLinearizedStresses) is that it can operate only on Mises, Tresca and major/intermediate/minor principal stress results plotted in the Line clip filter. However, it would be very useful to have a possibility to linearize the individual stress tensor components as well. They are available in the Line clip filter so it shouldn't be a problem to extend the functionality of the Stress linearization tool in this regard. Specifically, I'm talking about the following stress components:
- Stress xx component
- Stress xy component
- Stress xz component
- Stress yy component
- Stress yz component
- Stress zz component

Stress linearization is quite useful in the FEM industry (especially for pressure vessel design) and, from what I know, FreeCAD FEM is currently the only free FEA software capable of performing it.

I'm not familiar with the source code but wouldn't it be sufficient to add the aforementioned results variables to this condition in Command.cpp ?

```
if ((FieldName == "Tresca Stress") || (FieldName == "von Mises Stress")
            || (FieldName == "Major Principal Stress")
            || (FieldName == "Intermediate Principal Stress")
            || (FieldName == "Minor Principal Stress")
```

@marioalexis84 Can you comment on that ?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git)
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2023-12-18 by chennes

giorgiugo, 2025-03-28:
As I said in the forum I think that the calculation may be wrong. Is it possibile to check that before the next release?
Stress linearization is a calculation on stress components, after that is possible to calculate Von Mises of membrane and bending.
The code instead calculates a linearization of Von Mises.
https://forum.freecad.org/viewtopic.php?t=75602

FEA-eng, 2025-03-28:
@giorgiugo When I created this issue, it was indeed only possible to apply stress linearization directly to Mises, Tresca and principal stress results. But then I created a PR to close this issue and since then it is possible to perform stress linearization on individual stress components.

giorgiugo, 2025-03-29:
@FEA-eng Thank you for responding me. I'm not referring specifically to yor add. What I'm saying is that all tool is wrong.
Stress linearization (Asme VIII Div.2 5-A.4.1.2 or EN-13445) is a calculation on a tensor (all components in local coordinates referring to SCL: hoop, normal and tangential and not global because there's the need to take into account the right components and exclude some of them) and not on a vector (the code instead calculates formulas on an array (YAxis Data of the DataAlongLine that becomes sValues in the formulas)). An equivalent stress cannot be used as input, nor any other quantity taken individually beacuse the purpose is to calculate the tensor of membrane and bending and after that calculate the equivalent stresses of membrane, bending and membrane+bending in order to perfom the verifications.
So the inputs must be all the six stress components and x,y,z of the points of line (also to perform the coordinate transformation).
The outputs von Mises and Tresca of membrane, bending and membrane+bending and peak.
@luzpaz @chennes If i'm not misunderstanding the code I suggest freezing the tool until it is rewritten in order to prevent people from using it. What do you think about?

chennes, 2025-03-29:
@giorgiugo thanks for your detailed input. The best thing to do here is to create a new (unclosed) GitHub issue about the problem, and then we can ask @marioalexis84 to give his opinion on the proper course of action.

giorgiugo, 2025-03-30:
@chennes OK, I created #20522 . I hope it gets attention. Thanks.

PR #11724 (2023-12-18): Support for stress tensor components in FEM stress linearization
resolves #11719

Added support for stress tensor components in stress linearization. Tested on Ubuntu - works properly (it's a very small change anyway - just added 6 lines).