FreeCAD/FreeCAD #11720: [Problem] Themes other than Classic break the test_html_validates test

bgbsww, 2023-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you go to Edit -> Preferences -> General and choose a theme other than 'Classic'  - 'Dark' for example, and then either switch to the 'Test Framework' workbench, press the 'Self-test' button, and choose 'TestStartGui' and it will fail.

### Full version info

```shell
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35274 (Git) AppImage
Build type: Release
Branch: main
Hash: dc063eabec61bffc72c92595c66a4649f25fe381
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FreeCAD_BearingsWB 0.0.2
  * Assembly4 0.50.4
  * fcgear 1.0.0
  * Plot 2022.4.17
  * DynamicData 2.54.0
  * BIM 2021.12.0
  * parts_library
```

### Subproject(s) affected?

None

### Anything else?

To subsequently run from the command line:
C940:~$ ~/Software/FreeCAD_weekly-builds-35274-2023-12-03-conda-Linux-x86_64-py310.AppImage -t TestStart.TestStartPage

The 'Classic' theme appears to be the only one that doesn't reference a stylesheet, such as        
```
        <FCParamGroup Name="MainWindow">
          <FCBool Name="TiledBackground" Value="0"/>
          <FCText Name="StyleSheet">Light.qss</FCText>
          <FCText Name="OverlayActiveStyleSheet">Light_overlay.qss</FCText>
        </FCParamGroup>
```
which might be related.

There are a lot of errors generated, here's a few:

```
INFO: 'Trailing slash on void elements has no effect and interacts badly with unquoted attribute values.'
INFO: 'The \u201ctype\u201d attribute for the \u201cstyle\u201d element is not needed and should be omitted.'
ERROR: "CSS: \u201cselection-color\u201d: Property \u201cselection-color\u201d doesn't exist."
ERROR: "CSS: \u201cselection-background-color\u201d: Property \u201cselection-background-color\u201d doesn't exist."
ERROR: 'CSS: \u201cbackground-color\u201d: Parse Error.'
ERROR: 'CSS: \u201cbackground-image\u201d: \u201curl(qss:images_dark-light/background_freecad_dark.svg)\u201d is an incorrect URL.'

... much more of the same ...

INFO: 'Trailing slash on void elements has no effect and interacts badly with unquoted attribute values.'

----------------------------------------------------------------------
Ran 5 tests in 1.23s

FAILED (failures=1)

```

Here's a startpage.html that doesn't validate:

[freecad-startpage.zip](https://github.com/FreeCAD/FreeCAD/files/13662790/freecad-startpage.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-12-13:
It looks to me like all the stylesheets in src/Gui/Stylesheets have a large number of invalid tags set.  I hope these are autogenerated, because it's a big editing job if I'm right.

bgbsww, 2023-12-19:
In a brief Discord chat with @MisterMakerNL the suggestion was "the generator should only pick the colors. Leave out everything else.", but unfortunately the generator just swaps the stylesheet contents in wholesale: https://github.com/FreeCAD/FreeCAD/blob/976cd09b8ffe3aace9a5afbb0a99efde8d1a0c3a/src/Mod/Start/StartPage/StartPage.html#L12 via https://github.com/FreeCAD/FreeCAD/blob/976cd09b8ffe3aace9a5afbb0a99efde8d1a0c3a/src/Mod/Start/StartPage/StartPage.py#L468-L480

So that's somewhat non-trivial.  Is the right approach here to

1.   Stop trying to use an HTML validator on the startpage; disable or rewrite the test somehow.
2.   Make the generator code in startpage.py:handle somehow do something that allows the use of these .qss stylesheets  unmodified.  Given that some of the validation errors are on things like  `selection-background-color: @ThemeAccentColor1;`  I suspect an HTML validator will still fail.
3.   Change all the Stylesheet files to eliminate the bad lines.  I suspect this will break some of the intended purpose of the stylesheets.

It would be helpful if someone who knows the stylesheets and intent of this system could give some clues as to a direction forward here.

chennes, 2023-12-22:
I can tell you that I added that test because the Qt HTML rendering widget is (or at least _was_) notoriously sensitive to malformed HTML, so I wanted to ensure that whatever we fed it was 100% correct according to the standard. It's probably not a hugely important test to keep around, but it has caught various real errors in past PRs.

maxwxyz, 2024-08-03:
I hope you are ok when I close this as I cannot find this test anymore and I guess it was removed with the old Start WB

closed 2024-08-03 by maxwxyz