FreeCAD/FreeCAD #11721: [Problem] Crash when closing spline

Beep6581, 2023-12-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:
1. Start FreeCAD, Create New.
2. Sketcher.
3. Create Sketch, OK.
4. Create B-Spline, set mode to Periodic B-Spline By Knots.
5. Draw a square or a triangle. As you close the shape, it crashes.

[image]

```
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.

Thread 1 "freecad" received signal SIGABRT, Aborted.
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;

(gdb) bt full
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
        tid = <optimized out>
        ret = 0
        pd = <optimized out>
        old_mask = {__val = {93825057060664}}
        ret = <optimized out>
#1  0x00007ffff3eac8a3 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x00007ffff3e5c668 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
        ret = <optimized out>
#3  0x00007ffff3e444b8 in __GI_abort () at abort.c:79
        save_stage = 1
        act = {__sigaction_handler = {sa_handler = 0x20, sa_sigaction = 0x20}, sa_mask = {__val = {140733197470576, 1125, 140733197476048, 16, 93825059012680, 93825059012672, 93825059012632, 0, 140733196645362, 18446744073709551224, 2, 16, 140737285698387, 93825054837144, 93825054837320, 93825054837368}}, sa_flags = 1498431440, sa_restorer = 0x1f}
#4  0x00007ffff40dd3b2 in std::__glibcxx_assert_fail(char const*, int, char const*, char const*)
    (file=file@entry=0x7fff003e4b70 "/usr/include/c++/13.2.1/bits/stl_vector.h", line=line@entry=1125, function=function@entry=0x7fff003e60d0 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = double; _Alloc = std::allocator<double>; reference = double&; size_type = long unsigned int]", condition=condition@entry=0x7fff003dfada "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/debug.cc:61
#5  0x00007fff0032cba0 in std::vector<double, std::allocator<double> >::operator[](unsigned long) (__n=<optimized out>, this=<optimized out>) at /usr/include/c++/13.2.1/bits/stl_vector.h:989
        d = Python Exception <class 'gdb.error'>: value has been optimized out

        idxOfPole = <optimized out>
#6  GCS::BSpline::getLinCombFactor(double, unsigned long, unsigned long, unsigned int) (this=0x55555952b100, x=<optimized out>, k=<optimized out>, i=<optimized out>, p=<optimized out>) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/planegcs/Geo.cpp:859
        d = Python Exception <class 'gdb.error'>: value has been optimized out

        idxOfPole = <optimized out>
#7  0x00007fff00303520 in GCS::BSpline::getLinCombFactor(double, unsigned long, unsigned long) (i=5, k=<optimized out>, x=<optimized out>, this=0x55555952b100) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/planegcs/Geo.h:415
        i = 0
        numpoles = 2
        startpole = 5
        pvec = std::vector of length 1, capacity 1 = {0x5555594f7140}
        factors = std::vector of length 2, capacity 2 = {0.5, 0.5}
        constr = <optimized out>
#8  GCS::System::addConstraintInternalAlignmentKnotPoint(GCS::BSpline&, GCS::Point&, unsigned int, int, bool) (this=0x555559108998, b=..., p=..., knotindex=4, tagId=15, driving=true) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/planegcs/GCS.cpp:1493
        i = 0
        numpoles = 2
        startpole = 5
        pvec = std::vector of length 1, capacity 1 = {0x5555594f7140}
        factors = std::vector of length 2, capacity 2 = {0.5, 0.5}
        constr = <optimized out>
c
#9  0x00007fff00394f54 in Sketcher::Sketch::addInternalAlignmentKnotPoint(int, int, int) (this=0x555559108958, geoId1=<optimized out>, geoId2=10, knotindex=4) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/Sketch.cpp:3904
        p = <optimized out>
        b = <optimized out>
        tag = <optimized out>
        pointId1 = <optimized out>
#10 0x00007fff00395a93 in Sketcher::Sketch::addConstraint(Sketcher::Constraint const*) (this=this@entry=0x555559108958, constraint=0x555556054f00) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/Sketch.cpp:2246
        rtn = -1
        c = {constr = 0x555556054f00, driving = true, value = 0x0, secondvalue = 0x0}
#11 0x00007fff003963b1 in Sketcher::Sketch::addConstraints(std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> > const&, std::vector<bool, std::allocator<bool> > const&)
    (this=0x555559108958, ConstraintList=std::vector of length 15, capacity 15 = {...}, unenforceableConstraints=std::vector<bool> of length 15, capacity 64 = {...}) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/Sketch.cpp:2323
        it = 0x555556054f00
        rtn = <optimized out>
        cid = 14
#12 0x00007fff0038d8fc in Sketcher::Sketch::setUpSketch(std::vector<Part::Geometry*, std::allocator<Part::Geometry*> > const&, std::vector<Sketcher::Constraint*, std::allocator<Sketcher::Constraint*> > const&, int)
    (this=0x555559108958, GeoList=<optimized out>, ConstraintList=std::vector of length 15, capacity 15 = {...}, extGeoCount=<optimized out>) at /usr/src/debug/freecad/freecad/src/Mod/Sketcher/App/Sketch.cpp:252
        start_time = {_vptr.TimeInfo = 0x7ffff6244b50 <vtable for Base::TimeInfo+16>, timebuffer = {time = 1702485705, millitm = 506, timezone = 0, dstflag = 0}}
        intGeoList = std::vector of length 11, capacity 16 = {0x555557a809f0, 0x5555591c2ff0, 0x5555592c3a40, 0x555557a418d0, 0x5555593018b0, 0x55555917ceb0, 0x5555592b7800, 0x555559525640, 0x55555952d9e0, 0x555558a40280, 0x555559463ce0}
        extGeoList = std::vector of length 2, capacity 2 = {0x555559104500, 0x555558ffb370}
        onlyBlockedGeometry = std::vector<bool> of length 11, capacity 64 = {false, false, false, false, false, false, false, false, false, false, false}
        unenforceableConstraints = std::vector<bool> of length 15, capacity 64 = {false, false, false, false, false, false, false, false, false, false, false, false, false, false, false}
        blockedGeoIds = std::vector of length 0, capacity 0
        doesBlockAffectOtherConstraints = false
        extStart = <optimized out>
        extEnd = <optimized out>
```

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.71
  * Curves 0.6.15
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-12-13:
I cannot reproduce.

<details><summary>KUbuntu 22.04.3</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35335 (Git)
Build type: Debug
Branch: main
Hash: 0e5d2c46f0d2661d9f8b6692ae0bfad6c5504762
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

[image]

</details>

<details><summary>Windows 10 build 19044</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git)
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

[image]

</details>

christian-nils, 2023-12-14:
Cannot reproduce either on 

[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
[/code]

[image]

luzpaz, 2023-12-14:
Hello @Beep6581 nice to see the RawTherapee dev opening a ticket here. Welcome!

Out of curiosity can you reproduce on the appimage ? https://github.com/FreeCAD/FreeCAD/releases/tag/untagged-0f0a725fb4db2dfdd95e

Syres916, 2023-12-14:
I wasn't aware `OCC 7.7.2` was supported by FreeCAD yet, just because it can be compiled doesn't mean the core code has been enhanced to handle errors generated by it.

Anthony-Gaudino, 2023-12-17:
On 0.21.1, if I use Periodic B-Spline by knots and draw a square I have no crash, but it says `Solver failed to converge`.

Beep6581, 2023-12-19:
Hey @luzpaz 
Unfortunately that link does not work here, 404.

I tried https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/FreeCAD_weekly-builds-35414-2023-12-18-conda-Linux-x86_64-py310.AppImage and it works correctly - triangle, square and polygon.

luzpaz, 2023-12-20:
@Beep6581 thanks for confirming fix in v0.22.x

I'm not sure the fix will be backported to v0.21.x

closed 2023-12-20 by luzpaz