FreeCAD/FreeCAD #11725: [Problem] Sketcher automatic horizontal/vertical constraint is polemic and doesn't cover all usecases for individual tools

adrianinsaval, 2023-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

We've seen quite heated discussion about the introduction of this tool, the main problem as I see it is that sometimes a line is in a position where the tool would automatically place it as vertical but user actually desired a horizontal constraint (or vice versa).

Besides that this tool was introduced as a solution of the problem with constraining sketches that are rotated 90deg, while this solves the problem for mouse based constraining using keyboard shortcuts can still be frustrating.

### Full version info

```shell
OS: Arch Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35335 (Git)
Build type: None
Branch: main
Hash: 0e5d2c46f0d2661d9f8b6692ae0bfad6c5504762
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods: 
```
```

### Subproject(s) affected?

Sketcher

### Anything else?

I propose a few potential solutions to this:

• if a line already has a horizontal or vertical constraint and is clicked with this tool, the constraint is switched between them. This way if constraining a line doesn't give the desired result just clicking it again fixes this. This approach does nothing for the keyboard based constraining case and presents a problem when the line ends up being superposed with another line after applying the undesired constraint.

• Add a modifier key for the command, when holding alt (for example) the command could apply the opposite constraint that is auto detected, this way the user that realizes the line won't be constrained as desired can hold the modifier key. And if they didn't do that and get an undesired constraint they can undo and then reapply with the modifier. This somewhat addresses the keyboard problem if it's possible to call the command with the modifier directly. A tool tip when using this command would be required so that users actually learn about this possibility.

• Another option is to get rid of the combined auto tool and instead make the individual horizontal and vertical constraint commands smarter. Make them apply the opposite constraint if the view is rotated ~90deg. This also allows using familiar keyboard shortcuts when working with rotated sketches.

I'm gonna be honest, I almost never use the mouse commands for h/v constraints, I use the keyboard 99.9% of the time so option three would be the preferred solution for me as it would mean I can constrain as fast on a rotated sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-12-14:
thoughts @PLChris @PaddleStroke @abdullahtahiriyo

adrianinsaval, 2023-12-14:
About keyboard constraining, I see now that the combined tool has `A` as it's keyboard shortcut, this seems totally unintuitive to me. I would have expected `L` but this is taken by horizontal distance. So I also propose that if the combined tool is to be kept it's keyboard shortcut be changed to `L`, and horizontal distance should be changed to `-` which should be readily accessible on most if not all keyboards plus it's more visually similar (L was chosen for visual similarity initially, although it always seemed a poor choice to me), horizontal distance probably won't see much use with the existence of the new combined dimension tool anyways.

maxwxyz, 2023-12-14:
I really like the new implementation and never had any issues with it. If the desired constrain should be the other way around, you drew the base shape poorly, as it snaps to the nearest. Could you provide an example?
I like the A shortcut for align.

PaddleStroke, 2023-12-14:
1 - "A" was for align indeed. Thought it made sense but if you have better I have no strong feeling.
2 - Clicking on a vertical line could switch it to horizontal. It's a good idea, feels very convenient.
3 - The modifier is a good idea, but I feel that this kind of modifier are usually overlooked by most users. But for power users it sounds like a good idea.
4 - Removing auto tool and making the single tools sense the rotation sounds like a good idea. But I fear this may actually increase the confusion? As the tool would act differently based on something that is not directly related. I don't know.

Overall 4 may introduce more confusion so may not be worth it. 2 and 3 should be implemented. 
Then the question is about the default for which I have no strong feeling. Both situations have their advantages and disadvantages, but I think both options are acceptable. Putting the new tool as default during the dev cycle at least is a good idea because it will make users discover the tool and development version's goal should be that : experiment and get user's feedback on new ideas. People who cannot stand toolbars changing during dev cycle should not use dev version.

adrianinsaval, 2023-12-14:
> I really like the new implementation and never had any issues with it. If the desired constrain should be the other way around, you drew the base shape poorly, as it snaps to the nearest. Could you provide an example?
> I like the A shortcut for align.

https://forum.freecad.org/viewtopic.php?p=721421#p721421
and preemptively I quote chrisb from later on that thread:

> Please try to abstract from the simplified cases presented here. Of course I can get three lines right on the first try. But if I have more complicated sketches, things sum up, and in the end exactly this situation occurs in two or three places.

---

the word align is never mentioned in the tool, if it at least had that name it would be a more intuitive shortcut, maybe it could be renamed "Align to axis".

> 3 - The modifier is a good idea, but I feel that this kind of modifier are usually overlooked by most users. But for power users it sounds like a good idea.

I think it should have a very obvious tool tip while using the tool, like something just bellow the cursor.

> 4 - Removing auto tool and making the single tools sense the rotation sounds like a good idea. But I fear this may actually increase the confusion? As the tool would act differently based on something that is not directly related. I don't know.

I don't think it's confusing, when the sketch is rotated I think it's natural to expected such behavior. I'd like to hear other people's opinion on this, imagine for a second the new tool didn't exist (and don't think about the merits of that) just think if horz. and vert. constraints where auto inverted on a ~90deg rotated sketch, would you be happy?

The problem I see with this approach is that I don't know what the hell to do when the view isn't normal to the sketch, so perhaps it's better to just do 2. and 3. and call it a day.

PaddleStroke, 2023-12-14:
> on a ~90deg rotated sketch, would you be happy?

Yes when you rotated 90 it will be great. But what about being rotated at 80 then? So the answer to this is to use 45 as the limit. After 45 you switch to the other side. And this is where confusion can be imo. Imagine you rotated to 50, then it is likely the user will still expect horizontal to do a horizontal but he'll get a vertical. And then the case of not-normal views too.

maxwxyz, 2023-12-14:
As a bonus, maybe the sketch axis could display a small arrow with a H and V so you know how the sketch is oriented. Or also some indications when the grid is visible?
CATIA H/V axis: 
[image]

FEA-eng, 2023-12-14:
> As a bonus, maybe the sketch axis could display a small arrow with a H and V so you know how the sketch is oriented. Or also some indications when the grid is visible? CATIA H/V axis: 

I was also thinking about suggesting this before the auto vertical/horizontal tool was introduced. But I really like the new tool and would rather stay with it. Changing the meaning of separate vertical and horizontal tools would indeed cause a lot of a confusion so that's a big no for me. I guess that the complaints about the new tool are often related to the fact that people want or try to use it on lines that don't have near horizontal/vertical orientation but are close to 45 degrees. But then it just makes no sense to rely on such a tool and manual tools should be used. Other than that, I really see no issues with the idea behind the auto tool. Of course, there's still a room for some improvements like handling the aforementioned cases where a horizontal/vertical constraint is already applied to a line.

maxwxyz, 2023-12-14:
I am on your side, I really like the new tool. Users who don't like to use it or have edge cases where the tool itself would make no sense could still use the *old* H / V constraints and their shortcuts. I still would prefer on top  a little axis cross at the center of the sketch to visualize the sketch orientation as mentioned above.

kadet1090, 2023-12-16:
> Yes when you rotated 90 it will be great. But what about being rotated at 80 then? So the answer to this is to use 45 as the limit. After 45 you switch to the other side. And this is where confusion [...]

It should always be horizontal or vertical in regards to the view that user sees when entering the sketch. That's what user expects, anything else will be strange probably. 

Imho the separate tools should be fixed even if the automatic one would be here to stay, because this tool provides only automatic workaround not a proper solution. The automatic tool for me works well for lines, but it is not intuitive for aligning points (for example line ends), especially if points are close to each other. 

It'd also help if the icon by the cursor would change based on which geometry is hovered to tell user which constraint will be applied.

abdullahtahiriyo, 2023-12-17:
I must admit I am in the process of trying to figure out what to do with the new automatic tools and the traditional ones. My aim is to still offer both possibilities where it makes sense, and to decide on a case per case basis.

One think I do not think is right is the grouping of the "traditional" constraints below this automatic tool, but it is not the main complaint of this issue.

With automatic constraining tools, I think we might come to a different understanding if we do not call them constraints. Traditional constraints are constraints, because the outcome of the command is the given constraint. Automatic constraining tools are not constraints, are tools which apply some logic and generate one (or potentially more) constraints. With automatic constraining tools there is no limit to the considerations or output (you could train an AI on your brain EM energy to produce an output, fine with me). With Constraints, the output should, IMO be what one sees.

With this differentiation in mind, I tend to see several of the new tools (PoO and Coincident; H/V; Dimensioning) as automatic constraining tools. Some more rudimentary, others more elaborated. In fact, it may be possible to combine all three in a single tool, providing it with enough quality input (In fact, the Dimensioning tool does create some geometric constraints when changing mode). I think that for people enjoying automatic tools, the ideal is to have one tool to rule them all. So, I would hint towards development in this direction.

Regarding traditional tools, there are two groups of users, those who want to work this way, and others which want the space and prefer to get them out of the screen (perhaps, while still being able to count on their shortcuts sporadically).

Talking about this specific H/V automatic tool, I think it is a rudimentary precursor of an appropriate automated tool.
Indeed I tend to think in the line of @kadet1090, that the traditional tools may need fixing so that they work as expected for 90º rotated viewports. Then, the extra that the "Alignment constraining tool" (better right?) may provide is a tiny delta. 

In my current understanding, this tiny delta means that, either we do away with it, or we morph it into something providing more functionality (and justifying its existence).

Here I propose a two stage approach. First, see what we can do with the traditional constraints. Second, see what we can do with the automated alignment tool.

@PaddleStroke 

I know I have had this issue with H/V and the viewport, but I do not remember the specifics.

What do you think about what @kadet1090 proposes? Is it something you would be interested in working on?

kadet1090, 2023-12-17:
> One thing I do not think is right is the grouping of the "traditional" constraints below this automatic tool, but it is not the main complaint of this issue.

Why is that? The automatic tool combines the "traditional" ones into one tool, it makes sense to look for specific tooling under the auto one. This is quite a intuitive place to look for them, and people tend to click on dropdowns anyways to check what is available. But this will work only for tools that indeed combine abilities of tools that they hide. For example, if it was not possible to constraint diameter via the smart tool it should not be grouped under it.   

> With automatic constraining tools, I think we might come to a different understanding if we do not call them constraints. Traditional constraints are constraints, because the outcome of the command [...]

With that I totally agree, but it is the matter of semantics. I second @obelisk79 here - constraints are well, constraints that are part of the sketch. The *tool*bar contains tools for creating not constraints. Of course the simplest kind of tool is one that creates exactly one constraint but while such precision is sometimes wanted in most cases we should ease the process for the users if possible - hence the "auto" or rather combined tools. That's why I think that current arrangement (with dropdowns) is great - for the most cases you can rely on tool that works for broad range of cases and only care about specific one if you need it.

obelisk79, 2023-12-17:
Kadet stated my reasoning behind toolbar buttons perfectly. Regarding this issue, I think Adrian has a point, if anything, I would offer this:

1. The keyboard shortcuts for the old tools should remain. I find it odd that they stop working once removed from the toolbar definition. I think this may be a FreeCAD issue and not exclusive to sketcher.
2. I like the idea of using a modifier key. Alternatively, I am increasingly finding the 'tool settings' widget to be a viable option we should be leveraging in cases like these. Use the M key to cycle behavior between align, horizontal and vertical.

maxwxyz, 2023-12-17:
I agree with both points @obelisk79 the tool settings would be ideal!

PaddleStroke, 2023-12-18:
@abdullahtahiriyo I'm not sure to follow what you meant by Kadet's proposal. Did you meant Adrian's proposal? Actually I'm not sure to understand your reply @kadet1090 . When you enter a sketch horizontal isn't always horizontal (sounds like I'm missing something)?

> It should always be horizontal or vertical in regards to the view that user sees when entering the sketch.

Which seems to be against changing the behavior of horizontal/ver tools. But then you say : 

> Imho the separate tools should be fixed even if the automatic one would be here to stay, because this tool provides only automatic workaround not a proper solution. 

Which seems the be the opposit of previous statement. Unless there are cases where entering sketch the view is already 90 rotated?

kadet1090, 2023-12-18:
Hmm, I recall that I had a problem few times with sketches that are attached with 90 deg rotation around the Z-axis, which when edited was rotated and so "horizontal" and "vertical" were basically switched. I however cannot reproduce this right now and it seems that sketch rotates correctly to give user proper perspective. I don't think that this was changed recently, so maybe something with camera did not work properly in that cases and I was stuck with view rotated to model coordinate system. Nonetheless, if camera works fine then I feel that the constraints also work fine. In that case you are right that those two sentences are clearly contradicting - so please ignore them. 

However if user consciously rotate view the problem we still probably can do something about that. Adjusting the exact constraints to constraint based on the camera rotation is not well defined for angles other than 90 deg ones so it creates new problems to deal with. Maybe simple warning would suffice? 

Something like:
```
You are currently in rotated view of the sketch. In that situation horizontal 
and vertical constraints can yield unexpected and non-intuitive results. Like,
for example, horizontal constraint making line vertical on screen.  

[ ] Don't show that again                                               [ OK ]
```
 
Also some indication that we are rotated with respect to sketch coordinate system would be helpful as now it is hard to tell that we are dealing with rotated view and that can for sure cause some confusion.

yorikvanhavre, 2023-12-18:
> I like the idea of using a modifier key. Alternatively, I am increasingly finding the 'tool settings' widget to be a viable option we should be leveraging in cases like these. Use the M key to cycle behavior between align, horizontal and vertical.

+1

DrD4ffy, 2024-01-11:
Personally I like the new tool. For single ambiguous cases I can (and do) still press H or V, but most of the time I appreciate to pick one tool and then just click on a series of vertical and horizontal lines.