FreeCAD/FreeCAD #11728: [Problem] Once a FreeCad file is exported to *.step, the same *.step file cannot be opened in FreeCad.

PrimoLegionario, 2023-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I export the Freecad file saved in *.step format.
once the operation is finished, I try to open the new *.step file, but the software no longer responds. I try to do the same operation but "IMPORTING" THE FILE, but the result of fault it' same.
Important: the file generated by FreeCad in *.step format can be opened by other software such as CAD, however the image is dark monochromatic.

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-14:
did you try it in the latest stable or dev version of FreeCAD? Could you provide your FCstd and STEP file?

JohnOCFII, 2023-12-14:
> What is the latest version? The newest available version is 0.20.2

Latest stable version is 0.21.1 available here: https://www.freecad.org/downloads.php

Or the latest development version can be downloaded from GitHub here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds