FreeCAD/FreeCAD #11734: [Problem] Can't fill boundary of split curves from curves workbench

Anthony-Gaudino, 2023-12-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using split curves from the curves workbench to create an area to fill using the surface workbench, when trying to fill using `Fill boundary curves` or `Filling...` it fails to fill the area saying that `curves are disjoint`.

Please see example: [split_curve_bug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13677138/split_curve_bug.FCStd.zip)

### Full version info

```shell
OS: macOS 13.5.2
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.9.2
  * ThreadProfile 1.89.0
  * Launcher
  * InventorLoader 1.4.0
  * Render 2023.8.18
  * Silk 0.1.5
  * Defeaturing 1.2.1
  * A2plus 0.4.61
  * Manipulator 1.5.0
  * fasteners 0.4.71
  * lattice2 1.0.0
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.12
  * toSketch 1.0.1
  * Curves 0.6.15
  * Assembly4 0.50.6
  * Assembly3 0.12.0
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-15:
CC @tomate44 would you mind weighing in ?

tomate44, 2023-12-15:
This is not a FreeCAD bug. This issue should be closed.
The problem comes from the CurvesWB FreeHand_BSpline tool that had a default 1e-5 tolerance.
In latest [commit](https://github.com/tomate44/CurvesWB/commit/45b45a505af1d1cea52646bee475957bcfbe345f), the default tolerance has been lowered to 1e-7.
This tolerance value applies at creation time.
For existing FreeHand_BSpline objects, Tolerance property must be set using an expression.

closed 2023-12-15 by Roy-043