FreeCAD/FreeCAD #11735: [WIP] Design task: Add new Welcome Setup Tab to Start WB at first run

marcuspollio, 2023-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Problem

It has been pointed out in numerous occasions [1] that FreeCAD is not easy getting started, particularly for new or occasional users. One of the causes is the ability as a working tool environment to easily adapt to user requirements. [2]
Although it is preferable for the defaults to be sound and well thought-out, it will never be able to satisfy all users completely, as the range of possibilities is so extensive. One of FreeCAD's missions is to be accessible, flexible and customizable; as a community project, it does not intend to impose a particular workflow or to lock users. [3] In fact, numerous configuration options exist. Thus only the user can determine where the balance pointer between a default setting and a fully customized one is. The user can be assisted in this task.

One aspect that comes up regularly is the lack of user-friendly configuration of the application at start-up. [4-5] This serves as a gateway to facilitate the transition from installing the software to working on a project according to user requirements, with access to more if needed. This applies both to newcomers and frequent users who already have an previous version installed and can save time compared to diving into the myriad of options. One additional subsidiary benefit is to relieve the support platforms people of some repetitive queries from new users that can be easily answered with this new Setup.

This project is one of the goals for version 1.0 [6]

[1] Many posts of FreeCAD's Forum and other platforms, do we really need links ? =D
[2] FreeCAD Developers Handbook: https://freecad.github.io/DevelopersHandbook/roadmap/
[3] FreeCAD Website Homepage and Features: https://www.freecad.org
[4] Previous effort posted on the Forum: https://forum.freecad.org/viewtopic.php?t=71277
[5] Ondsel's Blog: https://ondsel.com/blog/freecad-needs-a-first-run-wizard/
[6] FreeCAD Developers Handbook: https://freecad.github.io/DevelopersHandbook/roadmap/next.html

Find below the related GitHub issues:

[7] https://github.com/FreeCAD/FreeCAD/issues/5772
[8] https://github.com/FreeCAD/FreeCAD/issues/10349
[9] https://github.com/FreeCAD/FreeCAD/issues/6106
[more maybe, please link them if you find any]

## Analysis and previously proposed solutions

Over the years, several attempts of a "First-start Wizard" (or other suggested names) have been made, as it seems indeed to be a viable solution to some of the problems mentioned above. However, none of the projects came to fruition due to lack of consensus on design or implementation difficulties. Ondsel then published a post highlighting the merits and pitfalls of the multiple works done and gave some recommendations. [5]

Find below the previous efforts:

[10] https://github.com/howie-j/freecad-wizard
[11] https://github.com/FreeCAD/FreeCAD-Getting-Started
[12] https://github.com/FreeCAD/FreeCAD/pull/10128

What emerges is the difficulty of designing and implementing this functionality, so simple and benign at first glance but ultimately crucial because it's visible to everyone from the very first launch and strongly influences the user's perception of FreeCAD's user-friendliness. What's more, these efforts have been led each time by a lone developer without a solidly validated design.

This new initiative was launched as part of the UI/UX working team as a community project bringing together several users and contributors, including some people involved in the previous efforts. It works like the recent successful changes to FreeCAD in successive phases (design, implement, test, refine).

Here's the first step which proposes a minimalistic Welcome Setup that covers the essentials: language, units, navigation and visual style (theme).

## Design

**Why?**

This new Welcome Setup answers genuine challenges as outlined above, while still being a moderate effort. Indeed, this implementation only takes care of certain aspects of the whole issue of ensuring that the configuration meets users' requirements, focusing on those with the most immediate impact. The presence of this Setup dialogue is kept to a minimum and not overly intrusive while guiding the user's first steps smoothly.

**What?**

A minimal Setup that appears the first time a new version of FreeCAD is launched. It integrates the Start Workbench with a new first Tab, it welcomes the user with a small text and proposes some configuration choices or to retrieve the configuration of the previously installed versions. After that, the Documents Tab is displayed. The following runs of the software will as it is now present the Documents Tab directly. The Setup Tab is still accessible if needed, just like the Help, Activity and Blog ones.

The Setup Tab is an interface with curated default settings options. They are first retrieved from the host system, which allows for example translation of the interface if a locale other than English is detected. When changed, these options override the ones in the Preferences and apply them immediately. When Preferences are changed, their new state is displayed in the Setup, recursively. It does not store Preferences Data, the current system in place for that is unchanged, for now.

**Where?**

The Start Workbench (introduced with version 0.12, improved in 0.18 and most recently in 0.22dev) serves a number of functions: opening recent documents, starting a new project with some pre-configured workbenches, seek help or learn new workflow tricks, access other FreeCAD's ecosystem resources and keep up-to-date with its development. This is the entry point after the splash screen.

As such, it makes sense to use it as the interface for the new Welcome Setup:
- some of the infrastructure is already in place and some contributors to the Start WB can be reached if needed
- it can be cleverly integrated using existing tabs without resorting to a pop-up style window
- its appearance can be easily customized

**How?**

The following mockups illustrate the current design proposed by the working group (credit to @qewer33) :

1) default in an English locale:

[image]

2) after changing the theme and the accent color:

[image]

**When?**

- First discuss, improve and validate this Design task
- Make sure the preliminary work is complete : fix #9297, configuration backup and easy retrieval system, themes are sufficiently mature to be integrated into the Setup, etc.
- PR for the implementation of the minimal Setup in following stages:
  - Add the new Tab but keep the current behaviour
  - Find a way to detect if a new version is installed and display the Setup in this case
  - Implement the button to retrieve previous configurations
  - Implement a functionnal Layout dropdown choice
- Test, adjust, retest, readjust, ... ok, merge !
- Define if other steps or adjustments are needed following broader user feedback
- Modify other aspects of the first run experience if necessary (Start WB and its other tabs, splash/launch screen, global defaults and Preferences, etc.)
- Champagne !

While it would be great to have all this in place for version 1.0, it's not sure it's going to play out smoothly.
A first PR prototype for Fosdem24 may be possible.

## Questions of Design or technical implementation to be answered before validation:

1) Should the Welcome Setup be displayed first when a new major/minor version changes, with point releases too ?
2) Interaction: when clicking on the link to open the Preferences window, should the Setup Tab be kept or the Documents Tab displayed instead ? 
3) More to be added...

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git) AppImage
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
```
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-12-15:
I like this mock up. Regarding question #1, I think displaying on major version isn't necessary. Otherwise the proposal is solid.

marcuspollio, 2023-12-15:
@obelisk79 Sorry the question wasn't clear, I reworded it. When a version change is detected, the Welcome Setup is opened at launch as the first Start WB tab (for following runs, the Documents Tab is opened like now). What should we consider as a "version change" ? Only between major/minor versions (from 0.21 to 1.0, 1.0 to 1.1, etc.) or also point releases (from 0.21.1 to 0.21.2, 1.0 to 1.0.1, etc.) ?

yorikvanhavre, 2023-12-15:
This is awesome! I also think integrating the first launch wizard to the start page is an excellent idea, because it allows us to use HTML, there are more fancy possibilities than Qt...  About the questions, my 2 cents:

1) I would say it should notify on all "published" releases: major and minor ones. The idea would be that people who "download" FreeCAD get notified, while people who use other channels (distro, PPA or self build), would not get annoyed, but that might prove difficult. In any case, it's good to know that a new release is out... Maybe we should just think of something discrete, that is  not annoying to have each time FreeCAD is opened
2) I think opening the prefs window should not change the tab, but no strong opinion on this...
3) For me yes it should skip the start page entirely... But, in that case, it still should show the welcome screen the next time FreeCAD is opened "normally" (without a file).

marcuspollio, 2023-12-17:
Hi @yorikvanhavre,

Thanks for your enthusiasm and answers !
1. Sorry I don't understand your answer. What do you mean by "get annoyed": show the Welcome Setup tab of the Start WB ? I'm not sure I get what you mean by "to know that a new release is out", somewhere else than the Splash screen and the About dialog ? The version info utility allows to find out how it has been installed, so which interaction do you propose in each situation?
2. Make sense.
3. Also make sens, to be seen how feasible it is.

FYI, I am off for the next weeks and be back early January. We can further discuss more in detail then, and hopefully get more feedback.
I wish you all wonderful winter/summer holidays !

yorikvanhavre, 2023-12-18:
By "get annoyed" I mean for ex. a message box that pops up. Not annoying would be a kind of exclamation mark or other sign next to the version number (or even the version number is in red and clickable... Something that users would not mind seeing everytime.

I'm thinking for ex, you are on ubuntu, and you get FreeCAD from the ubuntu repo. We publish a new version, but it'll take 6 months for that new version to land into the ubuntu repos. The user does want to keep the ubuntu version. He should not get annoyed by a message constantly reminding that his FreeCAD is not up to date for the next 6 months... That message could be there, but very unintrusive.

prokoudine, 2023-12-23:
1. The first-run wizard should only be displayed in two cases: 1) first installation in the system, 2) a new setting was added to the wizard and it requires user's attention, that usually happens in a major new release.

2. Keep the setup tab.

3. Let's say FreeCAD changes some defaults like the infamous Sketcher toolbar :) The user installs a new version, opens the file directly, FreeCAD skips the setup screen, and the user is then presented with an interface they didn't agree to. That would definitely cause friction. So I would suggest this: open the requested file, show the setup screen, then go immediately to the opened file after the user clicks "Get Started".

chennes, 2024-04-22:
I've started implementing this, but am a bit unsure about what the differences are meant to be between the different layouts ("classic", "modern", and "advanced") . Can you explain what preferences should be getting set by those?

qewer33, 2024-04-22:
- **Classic**: Same as now
- **Modern**: Tree View, Properties and Tasks View enabled as overlays along with tabbar WB selector
- **Advanced**: Similar to classic (but Tree View can be overlay maybe) but with more panels enabled (Report View, Python Console, Selection View etc.)

This is the general idea @chennes, it can be tweaked after the initial implementation.

katepavlu, 2024-05-28:
@maxwxyz an initial version of this is already implemented and discussion seems to have stopped. Should this remain open?

MisterMakerNL, 2024-05-28:
Layout part is still missing.
But @chennes hinted that it might be possible with preferences.

maxwxyz, 2024-07-11:
could this be closed now?

marcuspollio, 2024-07-16:
> could this be closed now?

I would hold off until #14694 is resolved.

maxwxyz, 2024-08-03:
I'll close this here then as #14694 is its own issue.

closed 2024-08-03 by maxwxyz