FreeCAD/FreeCAD #11737: AddOn-Manager was not able to load: Woodworking AddOn

Myszak64, 2023-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Response (in German):
"Laden der README-Datei von URL https://github.com/dprojects/Woodworking.git/blob/master/README.md ist fehlgeschlagen"

Installing other add-Ons was possible, but not the Woodworking stuff.

### Full version info

```shell
0.21.1
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-15:
For future reference, if you have a problem with an Addon _please_ raise it directly with the author, for example that would be at https://github.com/dprojects/Woodworking/issues but in this specific case I've raised a Pull Request with the author to hopefully fix the issue. The README.md not loading isn't a per-requisite to the Addon installing on **0.21.2** which is close to full release but it's not a professional look so it's now over to the author to merge my request, hopefully in the next few days assuming they are available.

Roy-043, 2023-12-15:
Closing this.

closed 2023-12-15 by Roy-043