FreeCAD/FreeCAD #11739: [Problem] Freecad default seems to be not template

amreo, 2023-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad says "template file is invalid" with default template files, so the default templates cannot be used (without modifications)

Steps to reproduce (with freecad flatpak)
1) Create a new project
2) Go to techdraw workbench
3) Create new techdrawing from template (TechDraw_PageTemplate) and select A4_Portrait_blank.svg (from .local/share/flatpak/app/org.freecadweb.FreeCAD/current/active/files/share/Mod/TechDraw/Templates/A4_Portrait_blank.svg)
4) Now freecad says that the "template file is invalid"

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at b9bfa5c)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * fasteners 0.4.66
  * 3DfindIT 1.2.0
  * freecad.gears 1.0.0
  * LCInterlocking
  * ModernUI (Disabled)
  * Beltrami 1.0.8-alpha
  * CurvedShapes 1.0.4
  * boltsfc 2022.11.5
  * ThreadProfile 1.87.0
  * sheetmetal 0.2.63
  * Curves 0.6.13
  * Assembly4 0.50.4
  * parts_library
[/code]
```

### Subproject(s) affected?

Techdraw

### Anything else?

I don't think I have changed techdraw settings.
The default templates seem to be here: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/TechDraw/Templates/A4_Landscape_blank.svg?short_path=cc20e1b

I have tried to generate a empty template using the first script in this page https://wiki.freecad.org/TechDraw_TemplateGenerator and the template works

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-16:
Your path contains the old FreeCAD URL. Are you sure that you are not using templates from some old version?

Try deleting the Template Directory in the TechDraw preferences, restart and try again.

amreo, 2023-12-16:
> Are you sure that you are not using templates from some old version?

I would say yes because I would have thought that .local/share/flatpak/app/org.freecadweb.FreeCAD/current/active/files/share/Mod/TechDraw/Templates/A4_Portrait_blank.svg corresponds to /app/share/Mod/TechDraw/Templates/A4_Portrait_blank.svg and that is provided by the flatpak package.

> Try deleting the Template Directory in the TechDraw preferences, restart and try again.

Instead of deleting directly the Template Directory I reset techdraw configuration because I had not set a template directory. Resetting the configuration worked for me.

So I think, this bug can be closed

closed 2023-12-18 by Roy-043