FreeCAD/FreeCAD #11741: [Problem] python module name of `MDIViewPy` derived windows is now `builtins`

Andrei-Pozolotin, 2023-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PROBLEM:

in recent revisions, the python module name of the `MDIViewPy` derived windows  has changed to the `builtins` 

VERIFICATION:

in python console:

to verify **proper class** behavior:
```python
main_window = FreeCADGui.getMainWindow()
this_type = type(main_window)
(this_type.__module__, this_type.__name__)
```
console output:
```text
('PySide2.QtWidgets', 'QMainWindow')
```

to verify **`MDIViewPy` class** behavior:
```python
main_window = FreeCADGui.getMainWindow()
active_window = main_window.getActiveWindow()
this_type = type(active_window)
(this_type.__module__, this_type.__name__)
```
console output for "Start Page":
```text
('builtins', 'BrowserView')
```
console output for "Spreadsheet":
```text
('builtins', 'SheetViewPy')
```
while previously that was:
```text
(`Spreadsheet`,`SheetViewPy`)
```

this is critical for type discovery, 
since  `MDIViewPy` classes do not expose `Module`, `TypeId` as is done in `FreeCAD.PyObjectBase`

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 (Git)
Build type: None
Branch: makepkg
Hash: ed10fe216d332debfd317d4639e284c50f5dab47
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-23:
What's the proposal for a fix then ?

Andrei-Pozolotin, 2025-02-03:
the issue is resolved as of:

```
OS: Arch Linux (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.40039 (Git)
Build type: Release
Branch: makepkg
Hash: 6c20224379d6f52dfbc768ebab4942f90bcb3ea4
Python 3.13.1, Qt 6.8.1, Coin 4.0.3, Vtk 9.4.1, OCC 7.8.1
```

closed 2025-02-03 by Andrei-Pozolotin