FreeCAD/FreeCAD #11748: [Problem] CSG renders improperly

dacto, 2023-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Opening a fairly simply CSG file (attached below; generated by OpenSCAD) with FreeCAD results in partial/incorrect rendering with extra/leftover(?) things.

The "Report View" logs (in additional info section below) reports `<Exception> Input shape is null` 4 times.

The OpenSCAD renderings shows the expected results.

OpenSCAD rendering:
Top:
[image]
Profile:
[image]

FreeCAD rendering:
Top:
[image]
Profile (and rotated to show the extra union better):
[image]

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 (Git) AppImage
Build type: Release
Branch: main
Hash: ed10fe216d332debfd317d4639e284c50f5dab47
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

OpenSCAD

### Anything else?

Input CSG file: [csg.zip](https://github.com/FreeCAD/FreeCAD/files/13694309/csg.zip)

FCStd file: [freecad_file.zip](https://github.com/FreeCAD/FreeCAD/files/13694318/freecad_file.zip)

Same incorrect result with FreeCAD 0.21.1.

---
Logs:
<details>

<summary>Report View Console Output</summary>

```
16:23:40  ImportCSG Version 0.6a
16:23:40  Start Lex
16:23:40  End Lex
16:23:40  Load Parser
16:23:40  WARNING: Token 'DOT' defined, but not used
16:23:40  WARNING: Token 'WORD' defined, but not used
16:23:40  WARNING: There are 2 unused tokens
16:23:40  Parser Loaded
16:23:40  Start Parser
16:23:40  Vector
16:23:40  Vector
16:23:40  Vector
16:23:40  Vector
[... 795 lines cut ...]
16:23:40  End processing CSG file
```
</details>

<details>
<summary>Python Console Output</summary>

```
Python 3.10.13 | packaged by conda-forge | (main, Oct 26 2023, 18:20:51) [GCC 12.3.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> ### Begin command Std_Open
>>> import importCSG
>>> importCSG.open(u"/work/misrendered.csg")
>>> # App.setActiveDocument("misrendered")
>>> # App.ActiveDocument=App.getDocument("misrendered")
>>> # Gui.ActiveDocument=Gui.getDocument("misrendered")
>>> # Gui.SendMsgToActiveView("ViewFit")
>>> ### End command Std_Open
>>> 
```
</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dacto, 2023-12-17:
Enabling the `Use Multmatrix Feature` option in FreeCAD's OpenSCAD settings resolved this issue for me.

I don't yet understand what this setting really does.
I'll continue digging, but so far I've only found:
* the brief tooltip
* https://wiki.freecad.org/OpenSCAD_Workbench with no mention of this
* https://wiki.freecad.org/OpenSCAD_Preferences which is just a screenshot of the UI
* a [post suggesting](https://forum.lulzbot.com/t/tip-converting-openscad-files-easily-to-step-with-freecad/228/2) it should be enabled

closed 2023-12-17 by dacto

KeithSloan, 2023-12-17:
It works with the Alternate Importer.

I don't know anything about a Multmatrix Feature option. Not sure how things would work without using Multmatrix facilities.

The Alternate importer as far as I know does not check for any such Feature Option.