FreeCAD/FreeCAD #11754: [Problem] python `Document.ActiveObject` does not reflect changes in `Model.TreeView`

Andrei-Pozolotin, 2023-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

PROBLEM:

`FreeCAD.activeDocument().ActiveObject` seems to reflect last `addObject(...)` 
and does not reflect actions/changes made by user in the `Model.TreeView` widget

VERIFICATION:

* create new document
* select `Spreadsheet` bench
* create multiple `Sheet`:  `Spreadsheet`, `Spreadsheet001`, `Spreadsheet002`

in python console:
```python
FreeCAD.activeDocument().ActiveObject.Name
```
output:
```text
'Spreadsheet002'
```

now in `Model.TreeView` double-click `Spreadsheet001`, and again:

in python console:
```python
FreeCAD.activeDocument().ActiveObject.Name
```
output:
```text
'Spreadsheet002'
```

while expecting: `Spreadsheet001`

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35405 (Git)
Build type: None
Branch: makepkg
Hash: 625a4212cc75b4fa7da6eb15f6538942290f8364
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-18:
This is not a bug but intended behavior.

To get the selected spreadsheet:
`Gui.Selection.getSelection()`

To get the opened spreadsheet:
`Gui.ActiveView.getSheet()`

Closing this.

closed 2023-12-18 by Roy-043

Andrei-Pozolotin, 2023-12-18:
OK. What is the meaning / purpose of `ActiveObject` intended behavior?

Roy-043, 2023-12-19:
As far as I understand: quick access to the last object. The name is indeed confusing. 

Related forum topic:
https://forum.freecad.org/viewtopic.php?t=80665

luzpaz, 2023-12-19:
Can't we link to documentation to clarify this ? Is there an .h or .xml file where this should be in ?