FreeCAD/FreeCAD #11759: [Problem] [TechDraw] Balloons may not be placed under Views in the TreeView

alexneufeld, 2023-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- Create a view of an object on a techdraw page.
- Select the view object's outline on the page.
- Use the add balloon command to add a balloon to the selected view by clicking on some geometry in the view.
- The balloon is **not** placed under the view in the TreeView.

based on my reading of the View's [Viewprovider's `ClaimChildren` method implementation](https://github.com/FreeCAD/FreeCAD/blob/625a4212cc75b4fa7da6eb15f6538942290f8364/src/Mod/TechDraw/Gui/ViewProviderViewPart.cpp#L238), this is a bug.

Example Screenshot:

[image]

See also the attached file.

### Full version info

```shell
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git) AppImage
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * FreeCAD-Flat-Icons
  * OpenDark 2023.12.1
  * TDAlignItems
  * fasteners 0.4.75
```
```

### Subproject(s) affected?

Techdraw

### Anything else?

[balloon_viewprovider_bug.zip](https://github.com/FreeCAD/FreeCAD/files/13700005/balloon_viewprovider_bug.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-18:
The balloon becomes a child of the view when the view is recomputed. Reopening the file has the same effect, so your example actually does not show the problem.

There is an issue here however. If you create a dimension it immediately becomes a child of the view. I do not know the TechDraw code, but I would think that the same should be done for balloons, hatches, etc.

WandererFan, 2024-02-14:
This is fixed in v0.22dev

closed 2024-02-14 by WandererFan