FreeCAD/FreeCAD #11760: Lines in concave corners can't be selected in 3D-view

Palmstroemen, 2023-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The yellow highlighted lines in the image below can't be selected. Just the neighbouring planes get highligted on mouseover.
[image]

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60h
  * CADExchanger
  * ExtremeProDark 2.7.0
  * fasteners 0.4.54
  * lattice2 1.0.0
  * MakerWorkbench
  * pluginloader
  * RemBench
  * sheetmetal 0.2.61
  * Silk 1.0.0
  * WebTools
  * workfeature
  * workfeature-macro
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-18:
This has been addressed and fixed in the current dev version.
Please retest with V0.22:
https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Closing this.

Palmstroemen, 2023-12-18:
Great. Thank you!