FreeCAD/FreeCAD #11761: [Problem] [sketcher] Buttons in the toolbar not very responsive since 0.21

Palmstroemen, 2023-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The buttons in the toolbar - especially for the constraints - are kind of 'not responsive' since the update to 0.21.
Sometimes, when I push a button it doesn't get activated. So I have to push it a second or even third time to get the tool activated.
I already got used to always push buttons twice.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.0.33668 +7 (Git)
Build type: Release
Branch: (HEAD detached at 0.21)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * A2plus 0.4.60h
  * CADExchanger
  * ExtremeProDark 2.7.0
  * fasteners 0.4.54
  * lattice2 1.0.0
  * MakerWorkbench
  * pluginloader
  * RemBench
  * sheetmetal 0.2.61
  * Silk 1.0.0
  * WebTools
  * workfeature
  * workfeature-macro
[/code]
```

### Subproject(s) affected?

None

### Anything else?

My subjective impression is, that FreeCAD did not get faster at all by 0.21 as promised. To me it seems to be slower and less responsive.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-18:
I can't reproduce this, either on V0.21 or V0.22. Are you using V0.21 on the same OS as the previous versions you are comparing to? Same mouse, same mouse driver?

maxwxyz, 2023-12-18:
Cannot reproduce. Did you try the latest dev builds 0.22?

Palmstroemen, 2023-12-18:
> I can't reproduce this, either on V0.21 or V0.22. Are you using V0.21 on the same OS as the previous versions you are comparing to? Same mouse, same mouse driver? If you are using a stylesheet: try switching that off.

I'm using V0.21 on the same Win11 system as before. Same mouse, same mouse driver (Laptop). I don't think that I'm using a stylesheet. I'm using FreeCAD out of the box.

Palmstroemen, 2023-12-18:
> I'm using V0.21 on the same Win11 system as before. Same mouse, same mouse driver (Laptop). I don't think that I'm using a stylesheet. I'm using FreeCAD out of the box.

No. I'm using V0.21 as mentioned above.

hasecilu, 2023-12-19:
Mmmm, I also had some problems in sketcher while trying to choose a secondary tool from their tool menu (Ex: Rounded rectangle, 3-points circle, etc). It was like if the tooltip was over the mini-menu and you couldn't click on the tool you wanted, initially I also thought it was my mouse or because of Hyprland.

That was on `0.21.1`, now I'm on `0.21.2` and I couldn't reproduce again. Arch btw.

FEA-eng, 2024-06-29:
@maxwxyz It seems that this can be closed as there's no sign it happens in 0.22.

closed 2024-06-30 by maxwxyz