FreeCAD/FreeCAD #11766: [Problem] Sketcher: UI - Task Dialog

maxwxyz, 2023-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The task dialog in sketcher is displaying constraints and geometry in two lists.
The constraints icons are huge in comparison to the geometry icons and the text seems not aligned.
Also the checkboxes seem different of the two lists. I am using the OpenDark theme but this is a general issue of the dialog.
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35407 (Git)
Build type: Release
Branch: sketcher_cursor_icons
Hash: 00aefca95bb2cca0f501e0cb5a92e75854f2a649
Python 3.11.7, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.16
  * Defeaturing 1.2.1
  * fasteners 0.4.74
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.1
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-19:
Checkboxes look the same here. I suspect that issues is related to the stylesheet.

I think the icons have (almost) the same size (15-16px), but the icons in the Elements list contain more details, and should therefore probably be bigger.

I don't see misalignment of texts.

maxwxyz, 2023-12-19:
@obelisk79 then it seems to be in the theme?

obelisk79, 2023-12-19:
Does the icon size change if you change theme?

maxwxyz, 2023-12-23:
it is also a different size and highlight color with other themes:
[image]

[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35449 (Git)
Build type: Release
Branch: main
Hash: f3fa2ba9c9b1999e5c32d3ae544481b6e17553bb
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3DfindIT 1.2.0
  * BIM 2021.12.0
  * CfdOF 1.24.9
  * CurvedShapes 1.0.5
  * Curves 0.6.19
  * Defeaturing 1.2.1
  * fasteners 0.4.75
  * FEMbyGEN 2.0.0
  * freecad.gears 1.0.0
  * freecad_metal_workbench 0.0.1
  * OpenDark 2023.12.17
  * sheetmetal 0.3.13
  * woodworking 0.21.1.33694
```

maxwxyz, 2024-01-07:
I'll close this as this is not a real bug but personal preference

obelisk79, 2024-01-17:
I'd reopen this, the issue of the different styled checkboxes is definitely real. The reason some people see check boxes that look the same in a classic or no-stylesheet theme configuration is because the element list checkboxes follow the system theme. So if your OS theme is light, and you apply a dark theme the checkboxes will mismatch.

PR #12054 (2024-01-23): Sketcher: Apply styling to elements and constraints UI
This ensures that checkboxes drawn in the Elements UI of the sketcher can be styled using the correct style.

Three changes are introduced:
1. paint() method uses style provided by the widget, not application.
2. Indicator is drawn using `PE_IndicatorCheckBox` not `CE_CheckBox` - it allows to use `::indicator` in qss to style it properly. This works the same as TreeView.
3. Fix for minor issues with text alignment and extension of selected items background to cover checkbox. This is how other controls of this kind work.

It also completely [... cut]
closed 2024-01-23 by chennes