FreeCAD/FreeCAD #11777: Sketcher radis constraint draws over circle arc

berberic2, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Sketcher radius constraint draws over the circle-arc it constraints. It did not do this in the past.
It’s distracting, annoying and absolutely useless.
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35391 +2 (Git)
Build type: Release
Branch: Meins
Hash: 576701c9356a4ffc5eadd19f6ef0015ee28b9e3a
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.15
  * Assembly4 0.50.4
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.3.12
  * Silk 0.1.5
  * CurvedShapes 1.0.5
  * DynamicData 2.54.0
  * FreeCAD_assembly3 0.11.3
  * fcgear 1.0.0
  * ExplodedAssembly
  * fasteners 0.4.72
  * Render 2023.8.18
  * A2plus
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berberic2, 2023-12-19:
The problem seems to shows only after at least one radius-constrain as reference is added. 
[radius_constraint.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/13720181/radius_constraint.FCStd.gz)

pierreporte, 2023-12-19:
Confirmed with the following version:
 
```
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35370 (Git) AppImage
Build type: Release
Branch: main
Hash: 1066bb54ceec8f4f8376b5149253acda5677651d
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

It appears for all kind of dimensions, not just references dimensions.

It looks like the dimensions it attached to one of the starting point instead of the arc itself in a similar way as lines, as reported in #11603.

berberic2, 2023-12-19:
Seems to already been fixed in 
```
Version: 0.22.0dev.35420 +2 (Git)
```

closed 2023-12-19 by berberic2