FreeCAD/FreeCAD #11778: [Problem] Sketcher_ConstrainLock works only for points

pierreporte, 2023-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:

* Select an entity (not a point), then Sketcher_ConstrainLock: FreeCAD displays an error message saying that the selection is invalid and that it should be a point different than the origin.
* Select Sketcher_ConstrainBlock then try to click on a point: the mouse pointer indicates an impossible action and the status bar says that the selection isn’t allowed.

Nothing but points can be blocked. It’s strange because this is the exact opposite of #11434 that was solved in the latest weekly I am testing with.

### Full version info

```shell
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git) AppImage
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * OpenDark 2023.12.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hasecilu, 2023-12-20:
`Sketcher_ConstrainLock` it supossed to work with points creating both a vertical and horizontal dimension constraints from the selected point to the origin.

hasecilu, 2023-12-20:
`Sketcher_ConstrainBlock` don't works if you select conflicting elements at the same time.

- If you have a rectangle and select 2 lines and then `Sketcher_ConstrainBlock` your drawing will be constrained
- If you have a rectangle and select 3/4 lines and then `Sketcher_ConstrainBlock` your drawing will be over-constrained and will led to a problem so it does nothing, still not smart enough to automagically discard extra constraints.

pierreporte, 2023-12-20:
@hasecilu I’ve seen the dimensions for the points. It’s great.

I tried to lock just one shape at a time (line, circle, spline, etc.) It never works if it’s not a point as explained in the opening comment. It only shows the following message:

> Wrong selection. Select one vertex from the sketch other than the origin.

hasecilu, 2023-12-20:
Mmm, at least with `Sketcher_ConstrainBlock` I'm able to block lines, circles splines... on 0.21.2 but I see other problems on 0.22.dev, Sketcher has had recent changes and it's expected to break.

maxwxyz, 2024-01-05:
I think this is not a bug. The lock is just working with points, for lines you have to constrain the start and end point or use the block constrain on the line.