FreeCAD/FreeCAD #11780: [PartDesign/Sprocket] Don't make dictionary keys translatable

hasecilu, 2023-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On Part Design WB you can create a sprocket, the Python code uses a dictionary to store the data and the keys to the dictionary are used for the user interface input and since were made translatable when the change it's done the keys don't match.

No problem with ANSI and ISO option but there are with bicycles and motorcycles options.

### Full version info

```shell
OS: LainOS (XFCE/openbox)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35420 (Git)
Build type: Unknown
Branch: main
Hash: 976cd09b8ffe3aace9a5afbb0a99efde8d1a0c3a
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Useless
  * FreeGrid
  * OpenDark 2023.12.1
  * Freecad-Built-in-themes-beta 1.2.2
  * A2plus 0.4.61
  * BIM 2021.12.0
  * woodworking 0.21.1.33694
  * Rocket 3.2.4
  * FreeCAD_SheetMetal 0.3.1
  * freecad_metal_workbench 0.0.1
  * kicadStepUpMod 10.22.4
  * Nodes 0.1.36
  * Defeaturing 1.2.1
  * MakerWorkbench 1.0.1
  * PieMenu 1.2.7
  * Help 1.0.3
  * FreeCAD_FastenersWB 0.4.75
  * cadquery_module
  * lattice2 1.0.0
  * Assembly4 0.50.6
  * Alternate_OpenSCAD 1.0.0
  * Assembly3 0.12.0
  * freecad.gears 1.1.0
  * Render 2023.8.18
```
```

### Subproject(s) affected?

PartDesign

### Anything else?

```error
17:54:48  Traceback (most recent call last):
17:54:48    File "/home/uli/git/github/FreeCAD_build_0.22/Mod/PartDesign/SprocketFeature.py", line 231, in sprocketReferenceChanged
17:54:48      self.obj.Pitch          = str(Sprocket.SprocketReferenceRollerTable[size][0]) + " in"
17:54:48                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
17:54:48  KeyError: 'Motocicleta 420'
18:01:35  Traceback (most recent call last):
18:01:35    File "/home/uli/git/github/FreeCAD_build_0.22/Mod/PartDesign/SprocketFeature.py", line 231, in sprocketReferenceChanged
18:01:35      self.obj.Pitch          = str(Sprocket.SprocketReferenceRollerTable[size][0]) + " in"
18:01:35                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
18:01:35  KeyError: 'Bicicleta con desviador'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2023-12-20:
@hasecilu if you want to try my fix, you can overwrite the current SprocketFeature.py with https://raw.githubusercontent.com/Syres916/FreeCAD/PD_Fix_Combobox_Issue/src/Mod/PartDesign/SprocketFeature.py

hasecilu, 2023-12-20:
Hi, I have tested your change and works perfectly fine, thanks.

I'm not very sure how strings are picked for translations, if I can recall correctly someone mentioned that there is a script to pick strings that were not explicitly called for translation, so if that's true  could be that how they were picked?

At least in external Python WB you have to use `tr("context", "translate this)` on every string but I don't know here.

closed 2024-01-07 by chennes