FreeCAD/FreeCAD #11781: [Problem] Axis too short

Beep6581, 2023-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Not sure if this is a bug or a user problem. I watched a youtube video where the horizontal and vertical axes cover the whole image:
https://youtu.be/CATOZ4mEwzo?t=407
In my case, the axes are too short, so I cannot anchor the b-spline to the vertical axis. In fact, sometimes they appear *after* I place a point, if I click beyond where they end.
[image]

https://github.com/FreeCAD/FreeCAD/assets/5844619/f9969810-9af3-404b-b0c0-b2399667f3ce

Is this by design?

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.11.6, Qt 5.15.11, Coin 4.0.1, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.75
  * CurvedShapes 1.0.5
  * Curves 0.6.19
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hasecilu, 2023-12-20:
Now the behaviour is "grow as needed" certainly a little annoying, you can always do the constraint after though

Beep6581, 2023-12-20:
I'm new to FreeCAD so I have no context, historical or otherwise, so with that in mind this feature seems a little like bloat to this newcomer... why maintain code for functionality that impedes design?

hasecilu, 2023-12-21:
As recent context the Sketcher WB has been going under a lot of changes recently.

Maybe we should ask to have the axis length big enough to fill all the 3D view, the reason is to have the axis available to make point constraints on them.

luzpaz, 2023-12-22:
@PaddleStroke can you weigh-in on this at your convenience ?

PaddleStroke, 2023-12-22:
Axis are indeed acting "grow on need" as was the grid before.
On one end it is annoying because as pointed out if you start already too
much zoomed out then you can't constraint to them while creating Geos.

We could change so that the axis are infinite, like the grid currently is.
I'd like to hear opinion from users about this. @Obelisk79 can you check
this in the design group?

Anyone see a problem if the axis were "infinite"?

JohnOCFII, 2023-12-22:
Infinite seems fine.

MisterMakerNL, 2023-12-22:
Normally in Solidworks I place an infinity construction line to do this.
But I have never tried it with infinite axis by default.
I would be a bit afraid if it would clutter the screen space a bit.
Like having a blank sketch space is nice to work with.
But I am not sure, maybe it is more convenient to have the infinity lines by default.

obelisk79, 2023-12-23:
@PaddleStroke after discussing it with the DWG members, no one saw any issues or had any objections with infinite axis lines.

hasecilu, 2023-12-23:
> Anyone see a problem if the axis were "infinite"?

The "infinity approach" is also used on guides on image programs like GIMP/Inkscape where it proves to be useful, in analogy in FC there is no such thing as infinite construction line as in SW (or does it?), so making the axis infinite is a `+inf`.

> Normally in Solidworks I place an infinity construction line to do this.

Would be nice to also have

maxwxyz, 2023-12-23:
@obelisk79 
If I could add another improvement regarding sketcher axes: 
The axes of the sketch should display a little arrow in edit mode (like the axis cross) to identify the orientation of the sketch easier (maybe with H/V indication). Currently there is no indication to inform the user if the sketch plane is viewed from the other side or in a rotated view and which is the horizontal or vertical direction.

Example of implementation in CATIA:
[image]

maxwxyz, 2024-04-12:
I guess this would also be solved by https://github.com/FreeCAD/FreeCAD/pull/12319

BernhardSchiffner, 2024-08-02:
Yes this is a double of #12319.
What I don't catch is:
These short axes are something new and are breaking user experience got with older (0.21) versions.
And I can't see an advantage of this.
IIUC it is told in #12319 discussion that short axes save some CPU-cycles. In my eyes endless axes they are worth it.