FreeCAD/FreeCAD #11783: [Problem] Access violation / Segmentation fault on Landmark Dimension

marzn, 2023-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

### Linux

If I open a file with a landmark dimension in a techdraw view, FreeCAD crashes with a segmentation fault. This was not present in FreeCAD 0.20.x, but the problem seems to exist since 0.21.0.

##### Steps to reproduce it
- Create a simple body
- Create two points
- Create a techdraw and a view
- Create a landmark dimension
- Save file
- Open file

A screenshot and a FCStd file of this minimal example is attached below.

##### CLI Output

```
FreeCAD 0.22.0, Libs: 0.22.0devR35420 (Git)
\u00a9 Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Adding snap-specific PYTHONPATH to sys.path: /home/boxes/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/870/lib/python3.10/site-packages:/snap/freecad/870/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
Loading Assembly workbench...
connect failed: No such file or directory
[18369:18375:1220/112455.124338:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
SE::getLocation3dFromFeat()
SE::getLocation3dFromFeat()
LD::execute() - LandmarkDim
LD::execute - features: 2
SE::getLocation3dFromFeat()
SE::getLocation3dFromFeat()
LD::execute - front:  (-34.15, 25.73, 0.00)  back:  (-34.15, 25.73, 0.00) 
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f43d504e520]
#1  /snap/freecad/870/usr/lib/TechDraw.so(+0x1f14e4) [0x7f42de72d4e4]
#2  0x7f42de72d998 in TechDraw::DrawViewPart::addReferencesToGeom() from /snap/freecad/870/usr/lib/TechDraw.so+0x1a8
#3  0x7f42de7853ff in TechDraw::LandmarkDimension::execute() from /snap/freecad/870/usr/lib/TechDraw.so+0xc7f
#4  0x7f43d73f273a in App::DocumentObject::recompute() from /snap/freecad/870/usr/lib/libFreeCADApp.so+0xba
#5  0x7f43d738d62d in App::Document::_recomputeFeature(App::DocumentObject*) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0x34d
#6  0x7f43d7395e8f in App::Document::recomputeFeature(App::DocumentObject*, bool) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0xef
#7  0x7f42de7057a3 in TechDraw::DrawPage::updateAllViews() from /snap/freecad/870/usr/lib/TechDraw.so+0xe3
#8  0x7f42de707ad8 in TechDraw::DrawPage::onDocumentRestored() from /snap/freecad/870/usr/lib/TechDraw.so+0x28
#9  0x7f43d7392b99 in App::Document::afterRestore(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0x6e9
#10  0x7f43d7393b51 in App::Document::afterRestore(bool) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0x51
#11  0x7f43d75f0dc0 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0x1d80
#12  0x7f43d75f1631 in App::Application::openDocument(char const*, bool) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0xf1
#13  0x7f43d760f60b in App::Application::sOpenDocument(_object*, _object*, _object*) from /snap/freecad/870/usr/lib/libFreeCADApp.so+0xfb
#14  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023) [0x7f43d69d7023]
#15  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7f43d698eadc]
#16  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7f43d692aa1c]
#17  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7f43d6a713af]
#18  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7f43d6a6c3de]
#19  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7f43d6abb4cd]
#20  /snap/freecad/870/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7f43d6abc4f9]
#21  0x7f43d6fd373c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/870/usr/lib/libFreeCADBase.so+0x6c
[... 78 lines cut ...]
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

Techdraw

### Anything else?

### Minimal example

I was able to create the following very simple drawing with FreeCAD on Linux. After saving, I was not able to reopen the file on Linux due to the segmentation fault discribed above. The landmark dimension is connected to DatumPoint and DatumPoint001.

![Screenshot](https://github.com/FreeCAD/FreeCAD/assets/13757952/3110a04c-b1c8-4b78-bee2-2166cdb20f77)

[demo2.FCStd.gz](https://github.com/FreeCAD/FreeCAD/files/13725828/demo2.FCStd.gz) (gzipped as FCStd is not allowed on GitHub)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hasecilu, 2023-12-20:
I tried your file but extracting it lead to an folder and changing extension to an error. Then I trie doing it myself but I don't see the datumPoint on the techdraw Page.
Could you upload a zip files?

marzn, 2023-12-20:
Thank you for investigating this.

The "DatumPoint" can be found in the "PartDesign" workbench, but I got the same result with the "Point" from the "Draft" workbench.

Here is a zip file containing the demo file from above: [demo2.zip](https://github.com/FreeCAD/FreeCAD/files/13733392/demo2.zip)

Is there anything else I can do to support here?

hasecilu, 2023-12-21:
> The "DatumPoint" can be found in the "PartDesign" workbench, but I got the same result with the "Point" from the "Draft" workbench.

What I meant it's that despite having the datum points they didn't appear on the Techdraw page and therefore I couldn't use the  landmark dimension tool on them.

[image]

Your new file is good and I can confirm that it instantly crashes FC when trying to open it.

With the info on https://wiki.freecad.org/Debugging I tried to debug FC 0.21.2 (1st time ever) it and the last output was:
```gdb
Downloading separate debug info for /usr/lib/freecad/lib/TechDrawGui.so
[New Thread 0x7ffed6f116c0 (LWP 336032)]
[New Thread 0x7ffed67106c0 (LWP 336033)]
[New Thread 0x7ffe7cbff6c0 (LWP 336034)]
[New Thread 0x7ffe77fff6c0 (LWP 336035)]
[New Thread 0x7ffe777fe6c0 (LWP 336036)]
[New Thread 0x7ffe76ffd6c0 (LWP 336037)]
[New Thread 0x7ffe767fc6c0 (LWP 336038)]
[New Thread 0x7ffed78886c0 (LWP 336039)]
[New Thread 0x7ffe75ffb6c0 (LWP 336107)]
LD::execute() - LandmarkDim
LD::execute - features: 2
LD::execute - front:  (-34.151, 25.727, 0.000)  back:  (-34.151, 25.727, 0.000)

Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >::operator=(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > > const&) [clone .isra.0] (this=this@entry=0x110, __x=std::vector of length 2, capacity 2 = {...}) at /usr/include/c++/13.2.1/bits/vector.tcc:236
236               if (__xlen > capacity())
(gdb) handle SIG33 noprint nostop
Signal        Stop      Print   Pass to program Description
SIG33         No        No      Yes             Real-time event 33
(gdb) bt
#0  std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >::operator=(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > > const&) [clone .isra.0] (this=this@entry=0x110, __x=std::vector of length 2, capacity 2 = {...}) at /usr/include/c++/13.2.1/bits/vector.tcc:236
#1  0x00007fff69169278 in TechDraw::GeometryObject::setVertexGeometry(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >)
    (newVerts=std::vector of length 2, capacity 2 = {...}, this=0x0) at /usr/src/debug/freecad/freecad/src/Mod/TechDraw/App/GeometryObject.h:115
#2  TechDraw::DrawViewPart::addReferencesToGeom() (this=<optimized out>) at /usr/src/debug/freecad/freecad/src/Mod/TechDraw/App/DrawViewPart.cpp:1355
[1]    326828 killed     gdb /usr/bin/freecad
```

It even frozed my computer on `bt` command for 9 minutes (bye audio server) :open_mouth: , I couldn't run `bt full`

> Is there anything else I can do to support here?

For now let's wait for a maintainer that tell us. Or do you know debugging? I don't really.

## Debug session
<details>
    <summary>Click to see the full gdb session</summary>

```gdb
$ gdb /usr/bin/freecad
GNU gdb (GDB) 13.2
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/freecad...

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) y
[... 600 lines cut ...]
[New Thread 0x7ffe77fff6c0 (LWP 336035)]
[New Thread 0x7ffe777fe6c0 (LWP 336036)]
[New Thread 0x7ffe76ffd6c0 (LWP 336037)]
[New Thread 0x7ffe767fc6c0 (LWP 336038)]
[New Thread 0x7ffed78886c0 (LWP 336039)]
[New Thread 0x7ffe75ffb6c0 (LWP 336107)]
LD::execute() - LandmarkDim
LD::execute - features: 2
LD::execute - front:  (-34.151, 25.727, 0.000)  back:  (-34.151, 25.727, 0.000)

Thread 1 "freecad" received signal SIGSEGV, Segmentation fault.
std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >::operator=(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > > const&) [clone .isra.0] (this=this@entry=0x110, __x=std::vector of length 2, capacity 2 = {...}) at /usr/include/c++/13.2.1/bits/vector.tcc:236
236               if (__xlen > capacity())
(gdb) handle SIG33 noprint nostop
Signal        Stop      Print   Pass to program Description
SIG33         No        No      Yes             Real-time event 33
(gdb) bt
#0  std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >::operator=(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > > const&) [clone .isra.0] (this=this@entry=0x110, __x=std::vector of length 2, capacity 2 = {...}) at /usr/include/c++/13.2.1/bits/vector.tcc:236
#1  0x00007fff69169278 in TechDraw::GeometryObject::setVertexGeometry(std::vector<std::shared_ptr<TechDraw::Vertex>, std::allocator<std::shared_ptr<TechDraw::Vertex> > >)
    (newVerts=std::vector of length 2, capacity 2 = {...}, this=0x0) at /usr/src/debug/freecad/freecad/src/Mod/TechDraw/App/GeometryObject.h:115
#2  TechDraw::DrawViewPart::addReferencesToGeom() (this=<optimized out>) at /usr/src/debug/freecad/freecad/src/Mod/TechDraw/App/DrawViewPart.cpp:1355
[1]    326828 killed     gdb /usr/bin/freecad
```

</details>

WandererFan, 2023-12-21:
Should be fixed by [TD]fix Landmark dim with datum points f9ea3bc

hasecilu, 2023-12-21:
I can confirm the file doesn't make FC to crash.
In the code I see `static bool isDatumPoint(App::DocumentObject* obj);`, can https://wiki.freecad.org/PartDesign_Point be added to:

> The TechDraw LandmarkDimension tool adds a linear dimension to a View. The dimension is based on two point objects ([Draft Point](https://wiki.freecad.org/Draft_Point) or [Part Point](https://wiki.freecad.org/Part_Point) objects) from the 3D model.

On https://wiki.freecad.org/TechDraw_LandmarkDimension or it's needed more changes?

WandererFan, 2023-12-21:
Good point.  Thanks for reminder.  TechDraw_LandmarkDimension has been updated.

luzpaz, 2023-12-23:
What's left here ?

closed 2024-02-14 by WandererFan