FreeCAD/FreeCAD #11789: Export to step fails.

Panda631, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I designed a part for a old radio model.

[image]

After finishing, I exported to step for 3D printing. No additional options clicked at the dialog box just OK. Than imported the step file back to FreeCAD (took a long time) and finaly looks like that:

[image]

The same step file imported to the slicer reported an error "non-manifold edges" and looks like that:

[image]

Can anyone comfirm that behavior? Maybe I did something wrong with the model. An export to stl works perfect!

Here the model.

[Step_problem.zip](https://github.com/FreeCAD/FreeCAD/files/13737845/Step_problem.zip)

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.0
  * Assembly4 0.50.4
  * EM
  * fasteners 0.4.70
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Manipulator 1.5.0
  * Render 2023.8.18
  * sheetmetal 0.2.61
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-22:
The cause of the problem was already found in the related forum topic:
https://forum.freecad.org/viewtopic.php?t=83681&start=10#p727218

Closing this as a user error.

closed 2023-12-22 by Roy-043

Panda631, 2023-12-22:
The descriped change does not solve the problem.!
The failure appears anyway.

luzpaz, 2023-12-22:
Please elaborate @Panda631

Panda631, 2023-12-22:
I can follow now Domad's solution.
Sorry for my fault and thanks for the investigation.