FreeCAD/FreeCAD #11790: [Problem] It is not possible to change the cutter diameter.

kharpost1, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For files created in Windows, when opening the file in Linux, you cannot change the diameter of the cutter.
This happens because the data about the cutter contains the path to the tool description file.
For example, this: C:/Program Files/FreeCAD 0.21/Mod\Path\Tools\Shape\endmill.fcstd
Naturally, such paths do not exist in Linux.

### Full version info

```shell
[code]
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.21.1.33668 +26 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-12-23:
Can you please provide an example file ?

kharpost1, 2023-12-29:
File created in Windows.
[bug_change_cuter_dia-win.zip](https://github.com/FreeCAD/FreeCAD/files/13794674/bug_change_cuter_dia-win.zip)

sliptonic, 2024-04-27:
this is normal behavior.  If the toolshape file isn't available, it isn't possible to alter its parameters.  The correct action would be to select a local toolshape for the toolbit.
Closing the issue since there's nothing to be done.

closed 2024-04-27 by sliptonic