FreeCAD/FreeCAD #11794: [Problem] Property view headers are useless.

PaddleStroke, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

The headers do not bring useful information as the view is self-explanatory. They just add a small clutter. Besides when used with overlay it's ugly : 

[image]

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2023-12-21:
When you do this there may also be an issue of the floating dots for the panel resize. Not just resizing the columns.

MisterMakerNL, 2023-12-30:
Totally agree with this!

closed 2024-01-10 by chennes