FreeCAD/FreeCAD #11799: [Problem] Sketcher : No visual feedback in sketcher to know if a wire is closed correctly.

PaddleStroke, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In sketcher you have no visual feedback to know if a wire is closed correctly. Which can lead to issue where the user thinks the sketch is closed. But one coincidence is actually missing.

There are possibilities that I can see to solve this : 
A - Draw un-coincident points with the construction geo color (blue now). And when points are coincident with another point they would become red. This way it would be very easy to see open wires.
If more than 3 points are stack, we just draw blue points on top of red ones.
[image]
This way it will be very obvious is a junction is not constrained.

B - Draw surfaces when a wire is closed (discussed here https://github.com/FreeCAD/FreeCAD/issues/11482)
With the possibility to show negative/positive surfaces.
[image]

Both proposal are not antinomic, and both could be implemented.

@abdullahtahiriyo @obelisk79 @maxwxyz 

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-12-21:
The islands are great but mainly for PartDesign features as @abdullahtahiriyo mentioned. I would like to have them as a toggle option on/off.
I propose three additional ways of indicating coincident points. If all points are coincident, the wire is closed.
- 1 Points are displayed hollow (only outline/stroke) if not at least two are coincident. They get filled when coincident (so a coincident sketch looks like a sketch today and only non-coincident points are hollow, or the other way around)
- 2 Coincident points are displayed with the current coincident icon
- 3 Coincident  points are displayed with another outline/stroke (used as coincident icon in most CAD software)
- 4 Display a small icon besides two points if they are close to each other but not coincident (not in the image below
- 5 Display the coincident constraint next to the points as any other constraint
- 6 Make all points which are not coincident squared (or the other way around)

[image]

abdullahtahiriyo, 2023-12-21:
I made some comments in the linked PR above regarding the solutions above.

This issue is presented as: "no visual feedback in sketcher to know if a wire is closed correctly". This wording points towards a subset of solutions, rather than the problem.

If the problem is "how do we identify mistakes associated with close points that are intended to be connected not being connected", perhaps there is a new range of possible solutions, which may not require to mark the coincident constraints.

After all, coincident constraints are but one of the possible ways to connect wires (end-to-end tangency being another one). Then, searching for red constraints (or their absence) is not very visual. From this point of view, Paddle's proposal (point changes colour) appears to be superior.

For Paddle's idea, I noted these drawbacks/notes:
i. There is an argument that it would be attributing yet another function to colour in the sketcher, as colour is already coding type of geometry, construction state and constrained state.
ii. I have some concern that some points will be red and others blue, without this meaning anything is inherently wrong or different (all are sketcher construction points with equal behaviour). I have some fear this will be misleading users, who would like to understand what is wrong, when nothing is inherently wrong with it.
(iii). On a separate point, moving construction points to blue by default would likely mean changing all art in the sketcher for consistency. Not a technical argument. Just a note.

Another approach (not a full solution, just another class of solutions)... Some time ago, I considered the possibility of running plausibility checks after each solve. These plausibility checks would be one or more of those in the sketch validation tools (open ends). This class of solution could mark on demand (for example green as in the trimming tool, or yellow with a similar circle marker) points that are close but unconnected. It is perhaps annoying when the user has two points that are close and not intended to be connected...

maxwxyz, 2023-12-21:
But a blue colored point would stay blue even when the point is fully constrained? So a open wire / path for surfaces, tool paths,... will always stay blue - if so, how does the user know these are constrained? Currently constrained lines and points are displayed green. If they will still be green, the other problem is the other way around - how to identify the open wire points? What about a square or diamond as a point display for open ends?

kadet1090, 2023-12-21:
I also think that we should not rely on introducing yet another color, sketcher even now has imho too much non-neutral colors. I think that changing shape for non-coincident (floating) points is fine - both hollow circle and other shape like diamond should be fine. However, hollow circle would by probably the easiest for users to understand. 

We still should however fill the shape - this would be the easiest thing to understand for users. I get that this is not always useful, but in context where such information is not needed you can simply ignore it or there should obviously be a way to turn it off.

PaddleStroke, 2023-12-21:
But we would not be introducing another color as it would use construction geo color.
Besides red points are construction geometries. So it would actually make more sense to have them blue right now. Because why points are red the color of constraints in the first place?

With my proposal construction points are blue like the rest of geos. And when they are coincident (or endpoint tangent) they become red, the color of constraints so that they indicate the constraints.

Now perhaps what you have in mind is that it is going to increase the overall 'rainbow' aspect of sketcher. But imagine we get construction geos to be dashed and gray. Then the points would simply be grey so it would also be much more neutral by default.

Regarding @maxwxyz proposals, which are basically all revolving around the use of shape rather than color I see 2 problems : 
- It actually adds one visual concept.
- It will not help for cases where you have 3 points, as the unconnected dot will be hidden by others.

Note an other alternative would be to have the point using the initial line color instead of construction geos color. So white lines would have white points, blue lines blue points. And once they are constrained they become red.

kadet1090, 2023-12-21:
Summarize of the discussion that we had on the topic on the DWG discord channel:

1. Closed shapes should be filled with semi-transparent background. This is standard behaviour in most of the CADs out there. This is a must, however it must be toggleable as not all cases require such information and it can get noisy especially for master sketches. Such toggle should be a sketch property if possible.

2. To give users even more information we can also mark "floating" points to be clearly visible where the problem is. So far we agreed that points should have two separate states:
    - Hollow meaning no coincidence is applied (via coincident. point on object, tangent or any other constraint) or Filled otherwise
    - Green meaning that the point is constrained, Red color meaning that point is not constrained (so just like it is now)

We also discussed possibility that points that are in the middle of geometry should be colored not as red but with the color of the line - so for example white for "normal" geometry and blue for constructive. This should make them not stand as much in the middle of path. I don't feel however that we achieved full agreement on this one.

Problem with points that has multi coincidence constraints applied (stars) was also mentioned as problematic as it is hard to say if two points that are part of paths are coincident or not:
[image]

As we need to balance there giving users the information that they need and not flooding them with too much information. Adding constraint icon to every point would result in sketches being way too noisy - which is a no-go. We discussed possibility of showing such icons only on hover so it'd be easy for users to check how many coincidences are applied to a given point. Other possibility proposed was to show the icon only if there are more than one coincidence constraints applied to a point. We did not come to a conclusion here yet.

PaddleStroke, 2023-12-22:
@kadet1090 PointOnObject should not be marked as 'full' in your example or that would defeat the whole purpose of showing open wires.

After sitting on it for a while, hollow circles seems like an acceptable solution to me, if the center is actually not transparent so that we can see unconnected points when 2 or more points are close.
 - But then I thought, why would we keep red as the base color of points? Points are construction geos so it would make more sense to be of construction color. Hollowed with blue points : 
[image]

- Then I thought, why keeping a point color alltogether? The points could be the color of their lines and be hollowed/not depending on their coincident status like this :
[image]

kadet1090, 2023-12-22:
> @kadet1090 PointOnObject should not be marked as 'full' in your example or that would defeat the whole purpose of showing open wires. I would say only coincident and end-point-tangency should be marker 'full'

If there is no case where point on object creates de facto coincidence you are absolutely right. 

As for the colors - I personally would get rid of them and implement them in a way that you show in your second picture. However I feel that red dots are part of the FreeCAD history and Identification so we should probably keep them at least partially.

About the blue points the above holds up and if we ever wanted to introduce construction points (for example because of #6884 which may require the distinction?) the color associated with construction should be left reserved.

PaddleStroke, 2023-12-22:
Ah I missed your meaning. Yes maybe 2 point on object can create a coincident. This  cases may be difficult to handle. It depends how we decide if points are coincident.

>  if we ever wanted to introduce construction points

Technically current red points ARE construction geometries. Create a point, it's red. Select it and toggle it with the construction toggle. It becomes white. White points are normal point geometries. But they are never used.

obelisk79, 2023-12-22:
What true use is there for distinguishing between construction point or not? Isn't a point just a coordinate position reference and not real geometry? The arc, line, spline segments are only what can be 'real' in sketcher from my understanding.

I initially thought blue vs white vs red would be good, but I am no longer convinced. Same can be said about the proposal of "hollow" point symbols. It is a good idea, unless point sizes get scaled down. Then the visual distinction from that loses value.

A red vs white color only approach 'may' be suitable but it should not introduce any new colors and still has problems which we discussed yesterday when more than two points are superimposed.

I need t think about this further to brainstorm if there are possible effective alternatives.

maxwxyz, 2023-12-22:
Not sure if other workbenches can work with the red / white dots (arch/path/draft/curves). Obviously we have a strong focus on the use of sketches in PartDesign.

obelisk79, 2023-12-22:
If anything the colors should be purely visual, only intended to convey information to the user. This would prevent unintended consequences within other workbenches. Any change in 'state' of a point I think is bound to cause problems.

pierreporte, 2024-01-05:
Is it necessary to have the visual feedback at all times? In other software, it is shown when using the sketch analyzer. You have a list of proximities and you can zoom in them.

> What true use is there for distinguishing between construction point or not? Isn't a point just a coordinate position reference and not real geometry? The arc, line, spline segments are only what can be 'real' in sketcher from my understanding.

Normal points can be useful for:

1. hole patterns (see #6884) and repetition patterns in general,
2. master sketches.

@PaddleStroke I agree with you that points could be always white. The normal points could be crosses and the midpoints a tick.

adrianinsaval, 2024-01-16:
> Ah I missed your meaning. Yes maybe 2 point on object can create a coincident. This cases may be difficult to handle. It depends how we decide if points are coincident.
> 
> > if we ever wanted to introduce construction points

While point on object could result in a coincidence situation this is not the proper way of making coincident point so I think it's ok to not mark those.

> Technically current red points ARE construction geometries. Create a point, it's red. Select it and toggle it with the construction toggle. It becomes white. White points are normal point geometries. But they are never used.

*rarely, they can be used to make lofts to point and I want to enable the hole tool to use them instead of circles too.

maxwxyz, 2024-01-17:
And points can be used in other WB without problems, there is not only part design.

maxwxyz, 2024-03-11:
There is also a good example from the forum: https://forum.freecad.org/viewtopic.php?p=746322

And it is also not possible to select a vertex which has 3 connected lines (star) and select the right coincident constraint and delete it. Because it is currently unclear which constraint is to which other line point.

Would it be possible to show the coincident constraints when the corresponding geometry is selected? Then the view is not cluttered by default and you could select all geometry and see which do not have a coincident constraint.

obelisk79, 2024-03-11:
I believe @kadet1090 had demonstrated the possibility of showing only geometric constraints for selected geometry a month or so ago. Not sure if he had plans to experiment further or draft a PR eventually.

closed 2024-06-04 by PaddleStroke