FreeCAD/FreeCAD #11801: [Problem] FreeCAD 21.1 Missing Default Plane in Polar Multitransform Tool

GITMuniac, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This problem surfaced first with the mirror transformation and was fixed.  See here ( https://github.com/FreeCAD/FreeCAD/issues/11571 ).  The same problem exists for Polar Transformations and I'm guess the fix is also just a matter of adding the code to supply a default plane (X/Y) like was done for Mirror.  Just wanted to post this as a reminder to the development team.  Many thanks for your efforts.

Best - Scott

### Full version info

```shell
OS: Linux Mint 20.3 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods:
* Assembly3 0.12.0
* sheetmetal 0.3.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2023-12-21:
Assign me, please.

GITMuniac, 2024-01-05:
Chennes - Thanks for taking care of this.  Happy New Year!

PR #11805 (2024-01-05): Provide a default axis for polar patterns in multitransformation
Fix #11801.