FreeCAD/FreeCAD #11807: [Problem] mouse overrides inputs in draft

kadet1090, 2023-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Various tools in draft allow both mouse and keyboard input, which is really nice. Keyboard input is preferred when precision is required, but then every move of the mouse above 3D view can reset values entered via keyboard, which can be really frustrating.

Some gifs illustrating the problem:
[image]
[image]

### Full version info

```shell
FreeCAD 0.22
```

### Subproject(s) affected?

Draft

### Anything else?

Mentioned by ProjectBP: https://forum.freecad.org/viewtopic.php?p=717666#p717666 - on discord we agreed that I'll create the issue on his behalf.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2023-12-22:
Possible approach to behavior change
(I apologize in advance for possible translation inaccuracies)
[image]

yorikvanhavre, 2023-12-22:
This is indeed something that bothers me as well. Not sure what would be a good solution here, because fundamentally it's hard to distinguish between when you are wanting or not wanting to have the mouse pointer updating the position.

Maybe this: When entering a character, you have a delay (one second? Definitely should be a preference option) during which the mouse input is ignored. So small unintentional mouse movements will be ignored while you are typing, but if you stop typing, the mouse takes over again

How does that sound?

kaiwas, 2023-12-22:
I have not seen such program behavior anywhere else.
If I enter data, it means that I need data and the mouse becomes a secondary auxiliary element. Now she's breaking everything.

Entering with the mouse "by eye" - enters virtually random positions (unless snapping to a grid can help out). This is only suitable for a rough sketch. But after the exact "dimensions", "dimensions", "dimensions"

I can give many examples from other programs where the mouse position does not spoil the data.
But there is not a single example where the approach is similar to that in freecad.

kaiwas, 2023-12-22:
Perhaps the checkbox is a non-standard solution and it would be possible to unblock the input fields using the escape key. But when pressed twice, it closes the tool.

However, the checkbox does not have such a problem. We freely move the mouse around the panel and, if necessary, enter data with the mouse, we activate it.

JohnOCFII, 2023-12-22:
Various task panels in `Path` workbench also suffer from errant mouse movements changing values in fields where the fields have not been actively selected.  Especially when using a TouchPad on a Macintosh laptop, this is incredibly frustrating. With other Apps on the Mac, I need to "double-click" into a cell to start changing values.  It seems many of the UI elements in FreeCAD don't require that step, causing to unexpected input.  Here's an example from Path that'll show changing values in fields without a double-click.
[image]

JohnOCFII, 2023-12-22:
By comparison - here's a preferences panel from Microsoft Excel.  I have to actively click into field to change the value away from the currently displayed  "10."   

[image]

kaiwas, 2023-12-23:
I had to install a demo version of Archicad from twenty years ago to show how the input fields work there.
Please pay attention. Until I started entering data into the fields - the data is taken from mouse movement. But as soon as I activated keyboard input, the mouse movement is completely ignored.
(I won’t talk about the convenience of entering directly in the 3D window, and not in the side panel. To do this, you need to create another topic. But nevertheless, all this can be seen in the GIF image)
[image]

yorikvanhavre, 2024-01-08:
I don't think we want to ignore mouse movements completely once a character has been entered, also because ATM you can enter "special" characters that are keyboard shortcuts for different functionality (snaps, close, etc...). So at the moment we still need both keyboard input and mouse input at the same time.

Let me try to introduce a delay and we see how that works...

kaiwas, 2024-01-08:
Can you give an example of at least one program where, after entering data, the mouse movement intercepts it? I haven’t seen anything like this, but it would be interesting to know that such things exist.

You can make an additional key for entering values using mouse movement. For example Alt+right mouse button.

A pause is, of course, better than nothing. Thanks for this try.
We could try it. But I repeat once again - the current situation as a whole is very inconvenient. There is no such behavior anywhere else!

For me, this was the first and most significant irritant because of which I postponed attempts to study freecad for several years.

In general, I understand the reason why you don’t want to change the usual order. But is it really necessary to give priority to the mouse over keyboard input? Does everyone really enter coordinates by moving the mouse “by eye”?

kaiwas, 2024-01-08:
> When entering a character, you have a delay (one second? Definitely should be a preference option) during which the mouse input is ignored. So small unintentional mouse movements will be ignored while you are typing, but if you stop typing, the mouse takes over again

It seems to be good. But this forces the user into a framework.
Imagine that I am building walls based on a reference. I need to enter the data, then look at the reference, find the right size there, and remember it. Then look back at the screen and enter the data.
All this may take much longer.
In the end, I can just think about the right size as much as I want.
At the same time, I have to remember about the time limit or just try not to breathe so as not to move the mouse.
The time limit is a half decision. In certain cases, this approach will be no less annoying than the current one.
However, it would be better than now.

yorikvanhavre, 2024-02-20:
If in the scenario above, you set the delay to, say, 1 hour, then it will have the effect you want. So I think it should be good for everybody

kaiwas, 2024-02-20:
> If in the scenario above, you set the delay to, say, 1 hour, then it will have the effect you want. So I think it should be good for everybody

Oh yeah. One hour should be enough.
If this is the easiest way then so be it. Thank you so much! That would truly be a victory.

But still, please note that the current behavior does not occur anywhere else (probably because it is really inconvenient).
But entering values using mouse movements could be done using some additional key. For example, Ctrl+holding the left mouse button (but I understand that this would depend on navigation styles).

yorikvanhavre, 2024-02-26:
Ok please test and play with PR #12624 ! I believe it solves the problem nicely.

kaiwas, 2024-02-27:
> Ok please test and play with PR #12624 ! I believe it solves the problem nicely.

Wonderful!
This is the best thing that happened this year so far!

kaiwas, 2024-03-02:
> Ok please test and play with PR #12624 ! I believe it solves the problem nicely.

How can I play with this?
I did not find the specified settings in the weekly build dated March 1st.

closed 2024-03-11 by chennes

kaiwas, 2024-03-19:
@chennes Hi.
They shouldn't have closed it. The problem is not solved.

yorikvanhavre, 2024-03-20:
I believe your version of FreeCAD does not have the fix yet. A way to verify is Edit -> PReferences -> Draft -> Grid and snapping and verify the "mouse delay" setting is there and its value is not zero

kaiwas, 2024-03-20:
> I believe your version of FreeCAD does not have the fix yet. A way to verify is Edit -> PReferences -> Draft -> Grid and snapping and verify the "mouse delay" setting is there and its value is not zero

Weekly-build version
 FreeCAD_weekly-builds-36487-2024-03-19-conda-Linux-x86_64-py311.AppImage 

In this build, this setting works for Draft tools (for example, move). But this doesn't work when entering wall lengths. And many other places.
In general, this should apply to absolutely all fields where the user enters data (has already entered it).

Rough input of coordinates using the mouse is only useful when snapping to grid is enabled. (In 25 years of work, I have never turned on the grid. There was no need for this) In other cases, the current behavior only introduces inconvenience.
Data is also lost in the sketcher.

PR #12624 (2024-03-11): Draft: Introduce mouse delay
This PR introduces a new "Mouse delay" preferences located under Edit -> Preferences -> Draft -> Grid and Snapping

* When mouse delay is set to 0, nothing happens, same as how it works now
* When it's set to a value, ex. 1 second, after you enter a number in one of the coordinate fields in Draft/Arch, the mouse will not change the value anymore during 1 second. After that, it resumes normal working.
* If you set the value to something very high, for ex. 3600 (one hour), mouse movement will be disabled after you start entering numeric values, until the command is f [... cut]
commit b2f1add7 (2024-03-11): Draft: Introduce mouse delay (#12624)

* Draft: Introduce mouse delay - fixes #11807

* Draft: set default mouse delay to 1