FreeCAD/FreeCAD #11810: expression using constraints in sketcher

jonascant, 2023-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Sketcher i want to reference the diameter of a circle to the length of a line using this formula: 400mm^2 * e^(2,3*Constraints.length_1)
This fails with "first arg must be quantity".
The Formula is parsed correctly if i use unit-less numbers only in the paranthesis.
The following attempt did not work: 400mm^2 * e^(2,3*Constraints.length_1/1mm)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (sway)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: German/Germany (de_DE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-22:
Confirmed.
Two variants that work:
`1 mm * 400 ^ 2 * pow(e; 2.3 * .Constraints.length_1 / 1 mm)`
`1 mm * 400 ^ 2 * e ^ 2.3 * .Constraints.length_1 / 1 mm`

pitch-circle, 2024-01-30:
I do not know if this is related but if you put 1^abs(-1) or 1^trunc(1.23) or 1^floor(18.7) into an Expression, FC fails consistently with the same message.

The workaround above, using pow(), is successful in all three cases.

Thus, it appears to be a problem with parsing what comes after the exponentiation operator rather than with united or unitless quantities.

If this is indeed a bug, then the Wiki page for Expressions needs a footnote for the ^ operator, saying that if it gives problems, use pow() instead.

FEA-eng, 2025-06-08:
Still relevant

FlachyJoe, 2025-06-09:
> `400mm^2 * e^(2,3*Constraints.length_1)`

 _first arg must be quantity_ is not the good message but IMHO the error is true: what should be the _1^mm_ unit?

> 400mm^2 * e^(2,3*Constraints.length_1/1mm)

 is _mm²_ so you can't set a length with this.

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```

maxwxyz, 2026-11-07:
Part of this is fixed in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/b2994fad4dd7ed547f504c3f897cc441f1429b1c

Dimensionless Quantity exponents work with numeric bases, including normalized constraint ratios. Exponents that still have units remain invalid.