FreeCAD/FreeCAD #11812: Clipping  Plane Dialog vanishes

calmarc, 2023-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I enable View->Clipping Plane. Then I select something and press onto the X to close 'dialog' there.
When I want to get rid of the clipping, I press again on 'View->Clipping Plane', but nothing happens. So I can not disable or change anything.
Unless I save and restart FreeCad.

### Full version info

```shell
OS: Ubuntu Noble Numbat (development branch)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git) AppImage
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.61
  * CurvedShapes 1.0.5
  * sheetmetal 0.3.12
  * fasteners 0.4.71
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

calmarc, 2023-12-22:
When it's in the transparent mode, I can get the 'dialog' back with Ctrl-Left etc. But somehow not when it's in within the solid grey on the left. (I also have no 'close' button there either, since there is not enough area for the whole 'Clipping Plane'-dialog

maxwxyz, 2024-02-22:
is this a duplicate: https://github.com/FreeCAD/FreeCAD/issues/11944

maxwxyz, 2024-02-22:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/11944

closed 2024-02-22 by maxwxyz