FreeCAD/FreeCAD #11816: Sketcher:  Distance constraint incorrectly drawn

MattN92705, 2023-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Sketcher occasionally draws "Distance" constraint dimension arrows in the incorrect place.  In the example below, a small notch is drawn in a square with a Distance constraint between a point and a line segment.  Sketcher draws the dimensional arrow from the point to the ORIGIN rather than to the line segment.  The actual constraint is correct, and the value shown in the dimensional arrow is correct.  Note the square is 30mm x 30mm, and the constraint 1mm.

[image]

Note that the same file opened in the release version of FC renders correctly:

[image]

### Full version info

```shell
OS: macOS 14.2.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35414 (Git)
Build type: Release
Branch: main
Hash: 759930e3aab711355d2f0895f641a8657897af48
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
```

### Subproject(s) affected?

Sketcher

### Anything else?

[DistanceBug.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/13759317/DistanceBug.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

howie-j, 2023-12-23:
I believe this bug is fixed in master by #11784 
The sketch renders correctly here:
[image]

```
OS: Fedora Linux 38 (Container Image) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35438 (Git)
Build type: Release
Branch: main
Hash: 6854b6d8ac0e94d6e02ebbcef2c7ccb4eeff5391
Python 3.11.6, Qt 5.15.11, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/United States (en_US)
```

MattN92705, 2023-12-23:
Indeed it seems to be fixed in 35449.  

Sorry to waste valuable bandwidth with this one.  I searched for the bug, but I guess my search terms missed it.

Carry on!

closed 2023-12-23 by MattN92705