FreeCAD/FreeCAD #11819: [Problem] crash when i edit the scale of a clone

drewp, 2023-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

* import STL
* Draft
* clone
* expand data->scale
* edit X from 1.0 to 2.0

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (fvwm3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33639 (Git)
Build type: Release
Branch: (HEAD detached at 1580fe7)
Hash: 1580fe7c9696654803a93da8cf372f75e75149fa
Python 3.10.11, Qt 5.15.10, Coin 4.0.0, Vtk 8.2.0, OCC 7.7.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Draft

### Anything else?

[image]

crash trace:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3f100) [0x7f575f03f100]
#1  0x7f56b7d860a2 in BRepTools_NurbsConvertModification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) from /app/lib/libTKBRep.so.7.7+0x1822
#2  0x7f56b7d7ecdd in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x1b8d
#3  0x7f56b7d7d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#4  0x7f56b7d7d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#5  0x7f56b7d7d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#6  0x7f56b7d7d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#7  0x7f56b7d800a7 in BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x357
#8  0x7f56b625b63e in BRepBuilderAPI_ModifyShape::DoModif() from /app/lib/libTKTopAlgo.so.7.7+0x6e
#9  0x7f56b625d76d in BRepBuilderAPI_NurbsConvert::Perform(TopoDS_Shape const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0x6d
#10  0x7f56b624ea2e in BRepBuilderAPI_GTransform::Perform(TopoDS_Shape const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0x4e
#11  0x7f56b624ee76 in BRepBuilderAPI_GTransform::BRepBuilderAPI_GTransform(TopoDS_Shape const&, gp_GTrsf const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0xf6
#12  0x7f56b92ff4c6 in Part::TopoShape::transformGShape(Base::Matrix4D const&, bool) const from /app/freecad/lib/Part.so+0xc6
#13  0x7f56b91955e7 in Part::TopoShapePy::transformGeometry(_object*) from /app/freecad/lib/Part.so+0xc7
#14  0x7f56b91956f2 in Part::TopoShapePy::staticCallback_transformGeometry(_object*, _object*) from /app/freecad/lib/Part.so+0x22
#15  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x126f48) [0x7f5760f26f48]
#16  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x9c) [0x7f5760f3961c]
#17  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x7898) [0x7f5760e754e8]
#18  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1ca405) [0x7f5760fca405]
#19  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x140f3e) [0x7f5760f40f3e]
#20  0x7f57619b112b in App::FeaturePythonImp::execute() from /app/freecad/lib/libFreeCADApp.so+0x19b
#21  0x7f56b90e5a6a in App::FeaturePythonT<Part::Feature>::execute() from /app/freecad/lib/Part.so+0x1a
#22  0x7f576190d09a in App::DocumentObject::recompute() from /app/freecad/lib/libFreeCADApp.so+0x9a
#23  0x7f56b90d91a0 in Part::Feature::recompute() from /app/freecad/lib/Part.so+0x10
#24  0x7f57618982dc in App::Document::_recomputeFeature(App::DocumentObject*) from /app/freecad/lib/libFreeCADApp.so+0x24c
#25  0x7f57618a1739 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /app/freecad/lib/libFreeCADApp.so+0x5c9
#26  0x7f576278955f in Gui::PropertyEditor::PropertyEditor::recomputeDocument(App::Document*) from /app/freecad/lib/libFreeCADGui.so+0x7f
#27  0x7f5762789634 in Gui::PropertyEditor::PropertyEditor::closeTransaction() from /app/freecad/lib/libFreeCADGui.so+0x94
#28  0x7f576278ca90 in Gui::PropertyEditor::PropertyEditor::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /app/freecad/lib/libFreeCADGui.so+0x50
#29  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x3062cc) [0x7f575fb062cc]
#30  0x7f5760a0ced9 in QAbstractItemDelegate::closeEditor(QWidget*, QAbstractItemDelegate::EndEditHint) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x59
#31  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x40d33c) [0x7f5760a0d33c]
#32  0x7f575faceb1b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x8b
#33  0x7f576077afee in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x6e
#34  0x7f5760782f1f in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xd4f
#35  0x7f5762439140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#36  0x7f575faced98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#37  /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1d9142) [0x7f57607d9142]
#38  0x7f576077affe in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#39  0x7f5762439140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#40  0x7f575faced98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#41  0x7f575ff52ef9 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5+0xe9
#42  /usr/lib/plugins/platforminputcontexts/libibusplatforminputcontextplugin.so(+0xc073) [0x7f575917a073]
#43  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x30604d) [0x7f575fb0604d]
#44  0x7f575a6405ff in QDBusPendingCallWatcher::finished(QDBusPendingCallWatcher*) from /usr/lib/x86_64-linux-gnu/libQt5DBus.so.5+0x3f
#45  0x7f575fafa9c0 in QObject::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x280
#46  0x7f576077affe in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7e
#47  0x7f5762439140 in Gui::GUIApplication::notify(QObject*, QEvent*) from /app/freecad/lib/libFreeCADGui.so+0x90
#48  0x7f575faced98 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x118
#49  0x7f575fad1d31 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x191
#50  /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x327403) [0x7f575fb27403]
#51  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a1) [0x7f575dd20d71]
#52  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xb0028) [0x7f575dd77028]
#53  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f575dd1e4a3]
#54  0x7f575fb26ae6 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x66
#55  0x7f575facd80b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#56  0x7f575fad5976 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x96
#57  0x7f5762365635 in Gui::Application::runApplication() from /app/freecad/lib/libFreeCADGui.so+0x2625
#58  FreeCAD(+0x910c) [0x55556c01310c]
#59  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x2958a) [0x7f575f02958a]
#60  /usr/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x7f575f02964b]
#61  FreeCAD(+0x9445) [0x55556c013445]
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-12-24:
In the image I do not see a mesh object. Can you attach the file? GitHub does not accept FCStd files but you can zip it.

drewp, 2023-12-26:
I couldn't exactly repro with the current state of the project, but here's a close repro using a .step file.

https://bigasterisk.com/post/freecad-11819/

```

FreeCAD 0.21.0, Libs: 0.21.0R33639 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Document with the UUID '95948c70-3e76-43d8-80a3-b009a277aec2' already exists, change to '87d87d3d-a039-4c5e-aed1-bb3d9d0baf97'
Attacher: linked object esp_resized is unexpected, assuming it has no shape.
Attacher: linked object esp_resized is unexpected, assuming it has no shape.
Attacher: linked object esp_resized is unexpected, assuming it has no shape.
Attacher: linked object Part is unexpected, assuming it has no shape.
Attacher: linked object esp_resized is unexpected, assuming it has no shape.
----------------
GuiCommand: Clone
Sketcher::SketchObject: Link(s) to object(s) 'Part Extrude' go out of the allowed scope 'Sketch013'. Instead, the linked object(s) reside within 'Part'.
Attacher: linked object Part is unexpected, assuming it has no shape.
Part::Extrusion: Link(s) to object(s) 'Sketch013' go out of the allowed scope 'Extrude008'. Instead, the linked object(s) reside within 'Part002'.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/x86_64-linux-gnu/libc.so.6(+0x3f100) [0x7fae1723f100]
#1  0x7fadff9860a2 in BRepTools_NurbsConvertModification::NewCurve2d(TopoDS_Edge const&, TopoDS_Face const&, TopoDS_Edge const&, TopoDS_Face const&, opencascade::handle<Geom2d_Curve>&, double&) from /app/lib/libTKBRep.so.7.7+0x1822
#2  0x7fadff97ecdd in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x1b8d
#3  0x7fadff97d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#4  0x7fadff97d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#5  0x7fadff97d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#6  0x7fadff97d679 in BRepTools_Modifier::Rebuild(TopoDS_Shape const&, opencascade::handle<BRepTools_Modification> const&, bool&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x529
#7  0x7fadff9800a7 in BRepTools_Modifier::Perform(opencascade::handle<BRepTools_Modification> const&, Message_ProgressRange const&) from /app/lib/libTKBRep.so.7.7+0x357
#8  0x7fad8285b63e in BRepBuilderAPI_ModifyShape::DoModif() from /app/lib/libTKTopAlgo.so.7.7+0x6e
#9  0x7fad8285d76d in BRepBuilderAPI_NurbsConvert::Perform(TopoDS_Shape const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0x6d
#10  0x7fad8284ea2e in BRepBuilderAPI_GTransform::Perform(TopoDS_Shape const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0x4e
#11  0x7fad8284ee76 in BRepBuilderAPI_GTransform::BRepBuilderAPI_GTransform(TopoDS_Shape const&, gp_GTrsf const&, bool) from /app/lib/libTKTopAlgo.so.7.7+0xf6
#12  0x7fadff6ff4c6 in Part::TopoShape::transformGShape(Base::Matrix4D const&, bool) const from /app/freecad/lib/Part.so+0xc6
#13  0x7fadff5955e7 in Part::TopoShapePy::transformGeometry(_object*) from /app/freecad/lib/Part.so+0xc7
#14  0x7fadff5956f2 in Part::TopoShapePy::staticCallback_transformGeometry(_object*, _object*) from /app/freecad/lib/Part.so+0x22
#15  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x126f48) [0x7fae19126f48]
#16  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x9c) [0x7fae1913961c]
#17  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x7898) [0x7fae190754e8]
#18  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1ca405) [0x7fae191ca405]
#19  /usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x140f3e) [0x7fae19140f3e]
#20  0x7fae19bb112b in App::FeaturePythonImp::execute() from /app/freecad/lib/libFreeCADApp.so+0x19b
#21  0x7fadff4e5a6a in App::FeaturePythonT<Part::Feature>::execute() from /app/freecad/lib/Part.so+0x1a
#22  0x7fae19b0d09a in App::DocumentObject::recompute() from /app/freecad/lib/libFreeCADApp.so+0x9a
...
```

Roy-043, 2023-12-28:
Duplicate of #9651
See https://github.com/FreeCAD/FreeCAD/issues/11834#issuecomment-1869122743
Closing this.

closed 2023-12-28 by Roy-043